ms-iot/ROSOnWindows #305: qt.qpa.plugin: Could not load the Qt platform plugin "windows" in "" even though it was found

WestbrookZero, 2020-12-19:
**Describe the bug**
qt.qpa.plugin: Could not load the Qt platform plugin "windows" in "" even though it was found.
This application failed to start because no Qt platform plugin could be initialized. Reinstalling the application may fix this problem.

Available platform plugins are: direct2d, minimal, offscreen, windows..

**Environment information:**
 - OS version: `win10`

seanyen, 2020-12-23:
@WestbrookZero Thanks for reporting this issue. Can you also specify what distro (e.g., Melodic, Noetic, or Foxy) and some steps how you run into this?

WestbrookZero, 2020-12-26:
Thanks for your answer! 
[image]
[image]
I use Melodic on win10, i set environment path :
Path
C:\opt\ros\melodic\x64\bin
QT_PLUGIN_PATH
C:\opt\ros\melodic\x64\plugins
QT_QPA_PLATFORM_PLUGIN_PATH
C:\opt\ros\melodic\x64\plugins\platforms
when i start "x64 Native Tools Command Prompt for VS 2019",Run as Administrator，as below：
cd C:\opt\ros\melodic\x64
setup.bat
rosrun rviz rviz
i still have the error: 
qt.qpa.plugin: Could not load the Qt platform plugin "windows" in "C:\opt\ros\melodic\x64\plugins\platforms;" even though it was found.
This application failed to start because no Qt platform plugin could be initialized. Reinstalling the application may fix this problem.

Available platform plugins are: direct2d, minimal, offscreen, windows.

IoTDan, 2022-10-13:
fixed in noetic.

closed 2022-10-13 by IoTDan