ms-iot/ROSOnWindows #308: Can't use Eigen?

qthqq, 2021-01-08:
I've installed the ROS Melodic on Windows binary using Chocolatey and can build basic ROS C++ nodes. Thanks for the work on this! 

However, if I try to include Eigen, it isn't found. I can verify that `Eigen/Core` seems to be in `C:\opt\ros\melodic\x64\include\eigen3`

My node is very simple. `mylib_hello_node.cpp`, in a package `mylib_gazebo` in a Catkin workspace. 

Builds and runs fine without `#include Eigen/Core`.

```
#include <ros/ros.h>
#include <Eigen/Core>

int main(int argc, char* argv[])
{
    ros::init(argc, argv, "mylib_node");

    ros::NodeHandle nh;

    ROS_INFO("Hello World!");
    ros::spin();
}
```

`catkin_make` fails with the following:

<details>

```
 [ 39%] Building CXX object mylib_gazebo/CMakeFiles/mylib_hello_node.dir/src/mylib_hello_node.cpp.obj
    mylib_hello_node.cpp
    C:\usr\x64\include\mylib-0.2-rc2\mylib/LibName.h(5): fatal error C1083: Cannot open include file: 'Eigen/Core': No such file or directory
    NMAKE : fatal error U1077: 'C:\PROGRA~2\MICROS~1\2019\COMMUN~1\VC\Tools\MSVC\1427~1.291\bin\Hostx64\x64\cl.exe' : return code '0x2'
    Stop.
    NMAKE : fatal error U1077: '"C:\Program Files (x86)\Microsoft Visual Studio\2019\Community\VC\Tools\MSVC\14.27.29110\bin\HostX64\x64\nmake.exe"' : return code '0x2'
    Stop.
    NMAKE : fatal error U1077: '"C:\Program Files (x86)\Microsoft Visual Studio\2019\Community\VC\Tools\MSVC\14.27.29110\bin\HostX64\x64\nmake.exe"' : return code '0x2'
    Stop.
    Invoking "nmake" failed
```

</details>

My `CMakeLists.txt`:

<details>

```

    cmake_minimum_required(VERSION 3.0.2)
    project(mylib_gazebo)

    ## Compile as C++11, supported in ROS Kinetic and newer
    ##add_compile_options(-std=c++11)

    ## Find catkin macros and libraries
    ## if COMPONENTS list like find_package(catkin REQUIRED COMPONENTS xyz)
    ## is used, also find other catkin packages
    #find_package(catkin REQUIRED)
    find_package(catkin REQUIRED COMPONENTS roscpp rospy std_msgs genmsg)

    find_package(eigen3 REQUIRED)
    find_package(Boost REQUIRED COMPONENTS chrono)

    #find_package(mylib REQUIRED)

    catkin_package(
    #  INCLUDE_DIRS include
    #  LIBRARIES mylib_gazebo
    #  CATKIN_DEPENDS other_catkin_pkg
    DEPENDS eigen3 Boost
    )

    ###########
    ## Build ##
    ###########

    ## Specify additional locations of header files
    ## Your package locations should be listed before other locations
    include_directories(
    # include
        ${catkin_INCLUDE_DIRS}
        #${mylib_INCLUDE_DIRS}
        ${Boost_INCLUDE_DIRS}
        ${eigen3_INCLUDE_DIRS}
    )

    add_executable(mylib_hello_node src/mylib_hello_node.cpp)

    target_link_libraries(mylib_hello_node
    ${catkin_LIBRARIES}
    #${mylib_LIBRARIES}
    ${Boost_LIBRARIES}
    )
```

</details>

There's some commented-out dependencies on `mylib` related to this [ROS Answers](https://answers.ros.org/question/369076/windows-dll-dependenciesexternal-non-ros/) question, but it seems like the problem is the Eigen dependency, which I think should be already installed in the default ROS Melodic installation?

qthqq, 2021-01-08:
In contrast, this works fine

```
#include <ros/ros.h>
#include <boost/chrono.hpp>

int main(int argc, char* argv[])
{
    ros::init(argc, argv, "mylib_node");

    ros::NodeHandle nh;
    boost::chrono::high_resolution_clock::time_point t0 = boost::chrono::high_resolution_clock::now();
    std::stringstream ss;
    ss << "Hello! The boost time was " << t0;
    ROS_INFO(ss.str().c_str());
    ros::spin();
}
```

```
C:\Code>rosrun mylib_gazebo mylib_hello_node
[ INFO] [1610067999.641345300]: Hello! The boost time was 716129781746900 nanoseconds since boot
```

qthqq, 2021-01-08:
Sorry, have confirmed this is a ROS/Eigen issue, nothing to do with Windows. Apparently Eigen CMake scripts use all-caps  `EIGEN3` prefix when setting things like `${EIGEN3_INCLUDE_DIRS}`

closed 2021-01-08 by qthqq

ooeygui, 2021-01-08:
@qthqq - Thank you for the investigation and follow up on this issue. 
We really appreciate it!