ms-iot/ROSOnWindows #309: Many Vcpkg ports are not usable in ROS on Windows installation.

seanyen, 2021-02-01:
Many Vcpkg ports are not usable in ROS on Windows installation. It failed at the following errors:

```
downloading bash-5.1.004-1-x86_64.pkg.tar.zst...
downloading filesystem-2021.01-1-x86_64.pkg.tar.zst...
downloading mintty-1~3.4.4-1-x86_64.pkg.tar.zst...
downloading msys2-runtime-3.1.7-4-x86_64.pkg.tar.xz...
downloading pacman-mirrors-20210127-1-any.pkg.tar.zst...
checking keyring...
checking package integrity...
loading package files...
error: could not open file /var/cache/pacman/pkg/bash-5.1.004-1-x86_64.pkg.tar.zst: Child process exited with status 127
error: could not open file /var/cache/pacman/pkg/filesystem-2021.01-1-x86_64.pkg.tar.zst: Child process exited with status 127
error: could not open file /var/cache/pacman/pkg/mintty-1~3.4.4-1-x86_64.pkg.tar.zst: Child process exited with status 127
error: could not open file /var/cache/pacman/pkg/pacman-mirrors-20210127-1-any.pkg.tar.zst: Child process exited with status 127
error: could not open file /var/cache/pacman/pkg/libzstd-1.4.8-1-x86_64.pkg.tar.zst: Child process exited with status 127
error: could not open file /var/cache/pacman/pkg/zstd-1.4.8-1-x86_64.pkg.tar.zst: Child process exited with status 127
error: could not open file /var/cache/pacman/pkg/pacman-5.2.2-11-x86_64.pkg.tar.zst: Child process exited with status 127
error: failed to commit transaction (cannot open package file)
```

The same failures can be reproducible when using the upstream Vcpkg `2020.07`.

seanyen, 2021-02-02:
Vcpkg `2020.11` and above doesn't have this problem. Most likely we need to upgrade Vcpkg to fix this issue...

bartholmberg, 2021-02-12:
I am trying to get rtabmap and/or cartographer_ros building on windows 10.  I see cartographer in the ROS share directory but can't build it from there.  Manual port from github repo is going better - so will probably continue there.