ms-iot/ROSOnWindows #310: ROS2 Development with Visual Studio

ItsMeTheBee, 2021-02-03:
Hey there!

I want to incorporate ROS into another Windows project where colcon is not an option. 
So I tried to write and run a simple C++ publisher in Visual Studio with a setup like [this one](https://answers.ros.org/question/340277/ros2-windows-running-c-node-by-direct-dlllib-reference/):

- C/C++ > Additional Include Directories = C:\dev\ros2_foxy\include
- Linker > Additional Library Directories = C:\dev\ros2_foxy\Lib
- Linker > Input > Additional Dependencies = all .lib files in C:\dev\ros2_foxy\Lib
- Copied all .dll files to output 

As far as I can see linking is not a problem, but there are two issues right now.
First problem is that I dont habe any pdb files, which means that I can´t debug into ROS. 
Would you happen to have these files at hand?

The second issue is that it seems like I get an `rclcpp::exceptions::RCLInvalidROSArgsError::RCLInvalidROSArgsError()` exception when calling the init function. I tried to call `rclcpp::init(0, nullptr);` and I tried to build argv myself to pass the path to the .exe file of the Visual Studio project or a working ROS node from my workspace but everything results in this exception.

I don´t expect you to know much about solving this second issue but maybe you happen to have a clue.

Best regards,
Sally

ooeygui, 2021-02-03:
Hi @ItsMeTheBee,
Thank you for your interest in ROS on Windows. We are always interested in talking with users of ROS on Windows and would love to know more about your project. Let me know if you are able to chat about it.

When building with cmake on Windows, pdbs are not typically generated for release builds. If you compile debug, you will have mismatches in the binaries. The ROS2 libraries you are importing into your C++ project were likely built without PDBs. If you need to debug ROS2 binaries, you'll need to rebuild them with `--cmake-args -DCMAKE_BUILD_TYPE=RelWithDebInfo`. 

For the second bug, I suspect that you maybe crossing toolchains. If you install ROS2 from the Microsoft chocolatey distribution (http://aka,ms/ros) these are built with the v142 toolchain. You'll need to make sure that is selected. (When building within the ROS environment, we take care of this for you)

If you can share a sample project, we can diagnose further.

Be well,
Lou

ItsMeTheBee, 2021-03-02:
Hey @ooeygui !

Thanks for your answer, I´ve been a bit busy but now I got some more infos.
I checked and the v142 toolchain is already selected.
I managed to build the debug binaries (just FYI I had some issues installing xmllint, not sure if it was my fault or if the tutorial for building ROS on windows is outdated but I needed to build zlib.lib myself).

The issues in `rclcpp::init(0, nullptr);` happened because I needed to add the path to the ROS binaries to my environment path.
Now I´m getting a "Invalid node name" error. I debugged around a bit and it seems like some characters are added which then lead to that error. Not sure where these characters come from, I added a screenshot of the contents of node_name when I build the class like this:  `MinimalPublisher():Node("")`.
[image]

I'm not sure how to share a sample project since all the ROS files (lib, dll, etc) are linked. I´ll try to push it into a new repo and send you the link.

Thanks a lot for your help!

ooeygui, 2021-03-02:
Thank you for the update and screen shots. I'm going to need to be able to reproduce this locally in order to help solve the mystery,

Typically when I see this, it is either a toolchain mismatch (which you ruled out), a calling convention problem (Which shouldn't be the case due to ROS2 code being explicitly tagged), or an optimization problem. The latter are really hard to diagnose because the generated code doesn't actually match what is written.

If you can step through the crossing, you can use the Memory and disassembly windows to see where things go off the rails. 

If you can't share your project, I can also do a one on one debugging session with you on your system if you are comfortable. Let me know, and I'll figure out how to contact you about setting that up.

ItsMeTheBee, 2021-03-03:
I think you can replicate the error with the code [here ](https://github.com/ItsMeTheBee/VisualStudioROS). The ReadMe should contain everything you need to know. 
If this doesn´t work we can do a debugging session as well =)

ItsMeTheBee, 2021-03-08:
Hahaha I feel kind of dumb for not trying this sooner, but I just changed the Solution Config to "Release" and it worked :D
Sometimes it's the simple things^^

ooeygui, 2021-03-08:
Ah, I should have had that on the list too - sorry about that. I'm glad it's resolved!

zanxueyan, 2024-05-17:
D:\ROS2_source>vcs import --input https://raw.githubusercontent.com/ros2/ros2/humble/ros2.repos src

Fatal error in launcher: Unable to create process using '"d:\python38\python.exe"  "D:\Python38\Scripts\vcs.exe" import --input https://raw.githubusercontent.com/ros2/ros2/humble/ros2.repos src'

What did I do wrong? How should I operate it?

ooeygui, 2024-05-17:
@zanxueyan, I'm not quite sure what you are trying to do, can you elaborate?
I'd recommend using the instructions at https://docs.ros.org/en/humble/Installation/Windows-Install-Binary.html.