ms-iot/ROSOnWindows #311: Saving Video and Screenshots from Gazebo on Windows?

danzimmerman, 2021-02-04:
Gazebo 9 shipped with ROS Melodic on Linux has a working mp4 video capture button built in.

This is not the case on my Windows installation of ROS Melodic binary install obtained via Chocolatey. Clicking the button does nothing.

[image]

I've been using Windows game bar thing (Windows +G ) to capture videos, but it's an awkward extra step and I feel like the video quality isn't as good as a natively saved file. 

I'm wondering if this is some relatively simple config thing that I can fix on my machine, like Gazebo is expecting to find some system installed `ffmpeg` or something?

The fact that the button doesn't respond makes it seem deeper than that, and now that I play around a little more it seems like the screenshot button crashes Gazebo as well.

ooeygui, 2021-02-04:
Thank you for the report.
Can you tell us more about your work with ROS on Windows?

danzimmerman, 2021-02-05:
>Can you tell us more about your work with ROS on Windows?

I'm working on developing a motion control library for a robot programming desktop software product. The product's cross-platform, but the main development track is Windows-centric, so I'm on Windows in Visual Studio most of the time. 

At the moment, my main goal with ROS is plugging in our library to a simulated environment for algorithm testing. (Grabbing simulated or real sensor inputs, operating in a dynamic world, developing some robot controller mocking). We've started building our library on Linux as well, so Windows usage isn't as critical as when I started, but it's definitely convenient. Also potentially interested in interop between ROS and our product for customers that might want it.

By the way, please let me know if this kind of `-desktop-full` minor-functionality bug report from outside your team and in the realm of my use case is useful or not to the goals of the project here. I do greatly appreciate the efforts to make ROS native to Windows, and so far it's very minor issues like this that have come up!