ms-iot/ROSOnWindows #312: ROS on windows.  What's new hasn't been updated since May 2020.

bartholmberg, 2021-02-12:
**Describe the bug**
ROS on windows , what is new page was last updated ===>> **May 2020**.   Given it was updated **monthly** prior to that, makes it appear ROS on windows is slightly dead :}

 https://ms-iot.github.io/ROSOnWindows/WhatsNew.html

 

**Expected behavior**
Update ROS on windows "entry page" each month [ even if nothing happened]

**Environment information:**
 - OS version: `winver`
 - Chocolatey Versions: `choco list --local-only`

ooeygui, 2021-02-12:
Thank you for the ping on that page. I'll update with back dated items.

bartholmberg, 2021-02-12:
good news!

ooeygui, 2021-02-12:
Here's the PR. There's some work that wasn't captured, but I need to collect that.
https://github.com/ms-iot/ROSOnWindows/pull/313

ooeygui, 2021-02-17:
Fixed. I'll make sure to keep it updated. Thanks for the ping on that!

closed 2021-02-17 by ooeygui