ms-iot/ROSOnWindows #315: Cannot open rviz on Windows 10. No errors on the console output.

vitoreno, 2021-02-14:
Hi all,
I report a bug I am experiencing on a fresh ROS noetic installation on Windows 10 (without WSL). I cannot open rviz, apparently no errors are reported on the console. I can open other qt based applications.

I wonder if someone could please help me solving this issue. Thank you

Best,
Vito Renò

**Describe the bug**
Cannot open rviz

**To Reproduce**
Steps to reproduce the behavior:
- Fresh install of ROS noetic from -> http://wiki.ros.org/noetic/Installation/Windows
chocolatey package ros-noetic-desktop_full
- execution of "roscore" in one administrator terminal
- execution of "rosrun rviz rviz" in a second administrator terminal
The following output is displayed:
`E:\> rosrun rviz rviz
[ INFO] [1613325936.385105300]: rviz version 1.14.4
[ INFO] [1613325936.385387200]: compiled against Qt version 5.15.0
[ INFO] [1613325936.385485500]: compiled against OGRE version 1.12.7 (Rhagorthua)
[ INFO] [1613325936.395256400]: Forcing OpenGl version 0.
E:\>`

**Windows event viewer**
Windows event viewer reports an error on rviz.exe executable

`Name of the application that generated the error: rviz.exe, version: 0.0.0.0, timestamp: 0x5fea7b78
Name of the module that generated the error: ucrtbase.dll, version: 10.0.18362.1110, timestamp: 0xb4cacc38
Exception code: 0xc0000409
Error offset 0x000000000006dace
Process ID that generated the error: 0x2da8
Start time of the application that generated the error: 0x01d702f9d40248b8
Path of the application that generated the error: c: \opt\ros\noetic\x64\Lib\rviz\rviz.exe
Path to the module that generated the error: C: \WINDOWS\System32\ucrtbase.dll
Report ID: 1fa7a184-05e5-47a4-9926-870ecc9633e5
Full package name that generated the error:
Application ID for the package that generated the error:`

**Expected behavior**
rviz should open

**Environment information:**
 - OS version: `winver`: windows 10 version 1909 build SO 18363.1379
 - set ROS output: 
`ROS_DISTRO=noetic
ROS_ETC_DIR=c:\opt\ros\noetic\x64\/etc/ros
ROS_IP=127.0.0.1
ROS_MASTER_URI=http://127.0.0.1:11311
ROS_PACKAGE_PATH=c:\opt\ros\noetic\x64\share
ROS_PYTHON_VERSION=3
ROS_ROOT=c:\opt\ros\noetic\x64\/share/ros
ROS_VERSION=1`
 -  set PATH output: `Path=C:\opt\ros\noetic\x64\tools\vcpkg\installed\x64-windows\bin;C:\opt\ros\noetic\x64\tools\vcpkg;c:\opt\ros\noetic\x64\Scripts;c:\opt\ros\noetic\x64\bin;c:\opt\ros\noetic\x64\; ...other non-ROS path directories...`
 - Chocolatey Versions: `choco list --local-only`:
`Chocolatey v0.10.15
chocolatey 0.10.15
chocolatey-core.extension 1.3.5.1
chocolatey-windowsupdate.extension 1.0.4
git 2.30.1
git.install 2.30.1
KB2919355 1.0.20160915
KB2919442 1.0.20160915
KB2999226 1.0.20181019
KB3033929 1.0.5
KB3035131 1.0.3
ros-noetic-desktop_full 20201214.0.0.2012281717
vcredist140 14.28.29325.2
vcredist2010 10.0.40219.2
13 packages installed.`

**Other information**
I am able to succesfully run rqt_gui or rqt_tf_tree

ooeygui, 2021-02-14:
Hi @vitoreno,
This is indeed very concerning, I'd like to help you get unblocked.
Can you post the command you used to launch the terminal window?
(I'm reinstalling noetic and will attempt to launch locally).

I know that this may seem random, but we have seen sensitivities to opengl.
What kind of video card are you using? Do you have the current drivers installed for it?

vitoreno, 2021-02-14:
Thank you! 

This is the command taken from settings.json of Windows Terminal
`{
    "commandline" : "C:\\Windows\\System32\\cmd.exe /k \"C:\\Program Files (x86)\\Microsoft Visual Studio\\2019\\Professional\\Common7\\Tools\\VsDevCmd.bat\" -arch=amd64 -host_arch=amd64 && set ChocolateyInstall=c:\\opt\\chocolatey&& c:\\opt\\ros\\noetic\\x64\\setup.bat",
    "guid" : "{f8e12390-4c29-4deb-a683-09567e2babea}",
    "icon" : "ms-appx:///ProfileIcons/{0caa0dad-35be-5f56-a8ff-afceeeaa6101}.png",
    "name" : "ROS Noetic",
    "startingDirectory" : "E:\\Software"
}`

My computer is an HP ZBook 15G3 with NVIDIA Quadro M1000M (driver version 27.21.14.5225 - 04/09/2020) and Intel HD Graphics 530 (driver version 27.20.100.8854 - 14/10/2020)

I don't know if it helps, but I also have a virtual machine installed on this computer, guest os Ubuntu 18.04 LTS and ROS melodic with working rviz

ooeygui, 2021-02-14:
Thank you for the info.  I'm not able to repro it on a reinstalled noetic.
If you are interested, I'd like to see if I can grab information off your system by building rviz and getting a stack trace. 
I'll post instructions for building when I get the minimal workspace.

ooeygui, 2021-02-14:
I need to follow up internally. 
In the mean time, can you create and attach a crash dump?

Launch roscore from one ROS terminal window.

Open a second ROS terminal window.
Then launch `devenv.exe` from that window (so it inherits the environment variables).
In Visual Studio, goto *File | Open* Project then type `c:\opt\ros\noetic\x64\bin\rviz.exe`
Then press play to debug.
When it crashes, Select *Debug | Save Dump As*
Attach it to this bug.

Thank you
Lou

vitoreno, 2021-02-14:
Please download the rviz dump at this link: https://cloud.ba.cnr.it/index.php/s/sGX2bArGZr7xpXE
The file size is too big to be attached here.

I also post here the output from debug
`
'rviz.exe' (Win32): Loaded 'C:\opt\ros\noetic\x64\bin\rviz.exe'. 
'rviz.exe' (Win32): Loaded 'C:\Windows\System32\ntdll.dll'. 
'rviz.exe' (Win32): Loaded 'C:\Windows\System32\kernel32.dll'. 
'rviz.exe' (Win32): Loaded 'C:\Windows\System32\KernelBase.dll'. 
'rviz.exe' (Win32): Loaded 'C:\Windows\System32\ucrtbase.dll'. 
'rviz.exe' (Win32): Loaded 'C:\opt\ros\noetic\x64\bin\msvcp140.dll'. 
'rviz.exe' (Win32): Loaded 'C:\opt\ros\noetic\x64\bin\vcruntime140.dll'. 
'rviz.exe' (Win32): Loaded 'C:\opt\ros\noetic\x64\bin\vcruntime140_1.dll'. 
'rviz.exe' (Win32): Unloaded 'C:\opt\ros\noetic\x64\bin\vcruntime140_1.dll'
'rviz.exe' (Win32): Loaded 'C:\opt\ros\noetic\x64\bin\vcruntime140_1.dll'. 
'rviz.exe' (Win32): Loaded 'C:\opt\ros\noetic\x64\bin\Qt5Widgets.dll'. Module was built without symbols.
'rviz.exe' (Win32): Loaded 'C:\Windows\System32\shell32.dll'. 
'rviz.exe' (Win32): Loaded 'C:\Windows\System32\cfgmgr32.dll'. 
'rviz.exe' (Win32): Loaded 'C:\Windows\System32\SHCore.dll'. 
'rviz.exe' (Win32): Loaded 'C:\Windows\System32\msvcrt.dll'. 
'rviz.exe' (Win32): Loaded 'C:\Windows\System32\rpcrt4.dll'. 
'rviz.exe' (Win32): Loaded 'C:\opt\ros\noetic\x64\bin\rviz.dll'. 
'rviz.exe' (Win32): Loaded 'C:\Windows\System32\combase.dll'. 
'rviz.exe' (Win32): Loaded 'C:\Windows\System32\bcryptprimitives.dll'. 
'rviz.exe' (Win32): Loaded 'C:\Windows\System32\windows.storage.dll'. 
'rviz.exe' (Win32): Loaded 'C:\Windows\System32\msvcp_win.dll'. 
'rviz.exe' (Win32): Loaded 'C:\Windows\System32\sechost.dll'. 
'rviz.exe' (Win32): Loaded 'C:\Windows\System32\uxtheme.dll'. 
'rviz.exe' (Win32): Loaded 'C:\Windows\System32\advapi32.dll'. 
'rviz.exe' (Win32): Loaded 'C:\Windows\System32\gdi32.dll'. 
'rviz.exe' (Win32): Loaded 'C:\Windows\System32\profapi.dll'. 
'rviz.exe' (Win32): Loaded 'C:\Windows\System32\win32u.dll'. 
'rviz.exe' (Win32): Loaded 'C:\Windows\System32\powrprof.dll'. 
'rviz.exe' (Win32): Loaded 'C:\Windows\System32\gdi32full.dll'. 
'rviz.exe' (Win32): Loaded 'C:\Windows\System32\umpdc.dll'. 
'rviz.exe' (Win32): Loaded 'C:\Windows\System32\user32.dll'. 
'rviz.exe' (Win32): Loaded 'C:\Windows\System32\shlwapi.dll'. 
'rviz.exe' (Win32): Loaded 'C:\Windows\System32\kernel.appcore.dll'. 
'rviz.exe' (Win32): Loaded 'C:\Windows\System32\dwmapi.dll'. 
'rviz.exe' (Win32): Loaded 'C:\Windows\System32\cryptsp.dll'. 
'rviz.exe' (Win32): Loaded 'C:\opt\ros\noetic\x64\bin\Qt5Core.dll'. Module was built without symbols.
'rviz.exe' (Win32): Loaded 'C:\Windows\System32\ole32.dll'. 
'rviz.exe' (Win32): Loaded 'C:\opt\ros\noetic\x64\bin\boost_filesystem-vc142-mt-x64-1_73.dll'. 
'rviz.exe' (Win32): Loaded 'C:\Windows\System32\ws2_32.dll'. 
'rviz.exe' (Win32): Loaded 'C:\opt\ros\noetic\x64\bin\boost_thread-vc142-mt-x64-1_73.dll'. 
'rviz.exe' (Win32): Loaded 'C:\opt\ros\noetic\x64\bin\Qt5Gui.dll'. Module was built without symbols.
'rviz.exe' (Win32): Loaded 'C:\opt\ros\noetic\x64\bin\resource_retriever.dll'. 
'rviz.exe' (Win32): Loaded 'C:\opt\ros\noetic\x64\bin\image_transport.dll'. 
'rviz.exe' (Win32): Loaded 'C:\opt\ros\noetic\x64\bin\boost_program_options-vc142-mt-x64-1_73.dll'. 
'rviz.exe' (Win32): Loaded 'C:\opt\ros\noetic\x64\bin\tf2_ros.dll'. 
'rviz.exe' (Win32): Loaded 'C:\opt\ros\noetic\x64\bin\message_filters.dll'. 
'rviz.exe' (Win32): Loaded 'C:\opt\ros\noetic\x64\bin\roslib.dll'. 
'rviz.exe' (Win32): Loaded 'C:\opt\ros\noetic\x64\bin\roscpp.dll'. 
'rviz.exe' (Win32): Loaded 'C:\opt\ros\noetic\x64\bin\class_loader.dll'. 
'rviz.exe' (Win32): Loaded 'C:\opt\ros\noetic\x64\bin\tf2.dll'. 
'rviz.exe' (Win32): Loaded 'C:\opt\ros\noetic\x64\bin\rosconsole.dll'. 
'rviz.exe' (Win32): Loaded 'C:\opt\ros\noetic\x64\bin\roscpp_serialization.dll'. 
'rviz.exe' (Win32): Loaded 'C:\opt\ros\noetic\x64\bin\tinyxml2.dll'. 
'rviz.exe' (Win32): Loaded 'C:\opt\ros\noetic\x64\bin\rostime.dll'. 
'rviz.exe' (Win32): Loaded 'C:\opt\ros\noetic\x64\bin\console_bridge.dll'. 
'rviz.exe' (Win32): Loaded 'C:\opt\ros\noetic\x64\bin\yaml-cpp.dll'. 
'rviz.exe' (Win32): Loaded 'C:\opt\ros\noetic\x64\bin\OgreOverlay.dll'. 
'rviz.exe' (Win32): Loaded 'C:\opt\ros\noetic\x64\bin\assimp-vc142-mt.dll'. 
'rviz.exe' (Win32): Loaded 'C:\Windows\System32\imm32.dll'. 
'rviz.exe' (Win32): Loaded 'C:\opt\ros\noetic\x64\bin\OgreMain.dll'. 
'rviz.exe' (Win32): Loaded 'C:\Windows\System32\mpr.dll'. 
'rviz.exe' (Win32): Loaded 'C:\opt\ros\noetic\x64\bin\msvcp140_1.dll'. 
'rviz.exe' (Win32): Loaded 'C:\Windows\System32\userenv.dll'. 
'rviz.exe' (Win32): Loaded 'C:\Windows\System32\version.dll'. 
'rviz.exe' (Win32): Loaded 'C:\opt\ros\noetic\x64\bin\zlib1.dll'. 
[... 47 lines cut ...]
'rviz.exe' (Win32): Loaded 'C:\opt\ros\noetic\x64\bin\libiconv.dll'. 
'rviz.exe' (Win32): Loaded 'C:\opt\ros\noetic\x64\bin\libintl.dll'. 
'rviz.exe' (Win32): Loaded 'C:\opt\ros\noetic\x64\bin\brotlicommon.dll'. 
'rviz.exe' (Win32): Loaded 'C:\opt\ros\noetic\x64\bin\libcharset.dll'. 
'rviz.exe' (Win32): Loaded 'C:\opt\ros\noetic\x64\bin\libcharset.dll'. 
'rviz.exe' (Win32): Unloaded 'C:\opt\ros\noetic\x64\bin\libcharset.dll'
'rviz.exe' (Win32): Loaded 'C:\opt\ros\noetic\x64\bin\icudt67.dll'. Module was built without symbols.
'rviz.exe' (Win32): Loaded 'C:\Windows\System32\secur32.dll'. 
'rviz.exe' (Win32): Loaded 'C:\Windows\System32\sspicli.dll'. 
[... cut ...]
'rviz.exe' (Win32): Loaded 'C:\Windows\System32\wshbth.dll'. 
'rviz.exe' (Win32): Loaded 'C:\Program Files\Bonjour\mdnsNSP.dll'. 
'rviz.exe' (Win32): Loaded 'C:\Windows\System32\DWrite.dll'. 
Exception thrown at 0x00007FFCC9C93B29 in rviz.exe: Microsoft C++ exception: boost::filesystem::filesystem_error at memory location 0x000000A708CFEE10.
Unhandled exception at 0x00007FFCC9C93B29 in rviz.exe: Microsoft C++ exception: boost::filesystem::filesystem_error at memory location 0x000000A708CFEE10.
`

ooeygui, 2021-02-14:
Thank you for posting. It looks like rViz is looking for a file - not a OpenGL problem like I thought. It must be an environment problem. I'll take a look at the dump you shared, and see If I can see which files it is attempting to load. (It does that quite a bit, so I'd have to narrow it down).

ooeygui, 2021-02-15:
Thank you for the dump file. (We do publish symbols for each build, I'll write some docs on how to load them and link back to sources).

The crash is occurring here:
```
 	[Inline Frame] rviz.dll!boost::filesystem::create_directory(const boost::filesystem::path &) Line 285	C++
>	rviz.dll!rviz::VisualizationFrame::initConfigs() Line 401	C++
 	rviz.dll!rviz::VisualizationFrame::initialize(const QString & display_config_file) Line 247	C++
 	rviz.dll!rviz::VisualizerApp::init(int argc, char * * argv) Line 210	C++
```

The dump doesn't have a memory dump, so I can't see what the value is. The crash is a codepath is trying to create the `.rviz` directory in your user account.

Some thoughts:
  * Your home directory is write protected?
  * Your user directory has characters that boost can't process?
  * Unlikely, but maybe your user directory is in a path that is too long?
  * Or maybe HOMEDRIVE and HOMEPATH aren't set?

vitoreno, 2021-02-15:
You were right at point number 2: my `HOMEPATH` actually is `\Users\Vito Renò` and it seems that boost does not like the `ò` character! I created a new administrator user (without any space or special character) on my computer and I am able to start rviz! 

Thank you very much for your help. I was not thinking about the accent in the HOMEPATH.

However, I would like to keep using my principal account, maybe changing the home directory or the user name/properties.
Do you know any safe procedure to do that? What you suggest to do?

Thanks again for the support!

ooeygui, 2021-02-16:
I've been doing some investigation. rViz needs a change to support international character sets on Windows. I'm adding an internal task to investigate support. 

I'm also following up on renaming the user account's directory. (It isn't as easy as it sounds), I want to make sure to share official guidance on this.

amelhassan, 2021-03-25:
Closing issue as this PR fixes it: https://github.com/ms-iot/rviz/pull/5.

closed 2021-03-25 by amelhassan