ms-iot/ROSOnWindows #316: tf2_geometry_msgs PyKDL error in NOETIC and ROS2 on Windows

MikeHallettUK, 2021-02-25:
I have tf2_geometry_msgs working fine on Windows MELODIC (Python 2) and Ubuntu NOETIC (Python 3).

But get this error on NOETIC Windows (and there is similar issue on ROS2 = see @ [https://github.com/ros2/geometry2/issues/285](https://github.com/ros2/geometry2/issues/285) ) 

Error = "PyKDL ModuleNotFoundError: No module named 'PyKDL' by importing import tf2_geometry_msgs"

Is it possible to get PyKDL for windows working with tf2_geometry_msgs for new ROS versions on Windows please ?

Thanks.

ooeygui, 2021-02-25:
Thank you for the report. I'll take a look at it.

While I can synthesize this locally, I want to make sure we address the issue you encountered. 
Could you point me at code online which encounters this error?

MikeHallettUK, 2021-02-26:
Just the import causes the error e.g. 

`from tf2_sensor_msgs.tf2_sensor_msgs import PointCloud2, do_transform_cloud`

You can get a sample program using this @ [camtfpc.py](https://github.com/MikeHallettUK/RosRobotics/blob/master/camtfpc.py)

Error msg:

> Traceback (most recent call last):
>   File "C:\ws\turtlebot3\devel\lib\mytbpc\mylistener.py", line 15, in <module>
>     exec(compile(fh.read(), python_script, 'exec'), context)
>   File "C:/ws/turtlebot3/src/mytbpc/mylistener.py", line 6, in <module>
>     from tf2_sensor_msgs.tf2_sensor_msgs import PointCloud2, do_transform_cloud      # to support PointCloud2
>   File "C:\opt\ros\noetic\x64\lib\site-packages\tf2_sensor_msgs\__init__.py", line 1, in <module>
>     from .tf2_sensor_msgs import *
>   File "C:\opt\ros\noetic\x64\lib\site-packages\tf2_sensor_msgs\tf2_sensor_msgs.py", line 30, in <module>
>     import PyKDL
> ModuleNotFoundError: No module named 'PyKDL'

The ROS noetic is installed from choco e.g. 

**C:\ws>choco upgrade ros-noetic-desktop_full -y --execution-timeout=0
Chocolatey v0.10.15
Upgrading the following packages:
ros-noetic-desktop_full
By upgrading you accept licenses for the packages.
ros-noetic-desktop_full v20201214.0.0.2012281717 is the latest version available based on your source(s).**

closed 2022-07-20 by MikeHallettUK