ms-iot/ROSOnWindows #317: ROS cv_bridge getCvType ImportError: DLL load failed while importing cv_bridge_boost in NOETIC

MikeHallettUK, 2021-02-28:
**Describe the bug**
getCvType ImportError: DLL load failed while importing cv_bridge_boost in NOETIC on Windows10:  this otherwise works fine on UBUNTU 20.04

>   File "C:\opt\ros\noetic\x64\lib\site-packages\cv_bridge\core.py", line 99, in encoding_to_dtype_with_channels
    return self.cvtype2_to_dtype_with_channels(self.encoding_to_cvtype2(encoding))
  File "C:\opt\ros\noetic\x64\lib\site-packages\cv_bridge\core.py", line 91, in encoding_to_cvtype2
    from cv_bridge.boost.cv_bridge_boost import getCvType
ImportError: DLL load failed while importing cv_bridge_boost: The specified procedure could not be found.

**To Reproduce**
Try code sample @ [RosViewer.py](https://github.com/MikeHallettUK/RosRobotics/blob/master/RosViewer.py)

**Expected behavior**
Program node listens to a ROS image message topic, and displays the image using OpenCV.
In sample RosViewer.py, line 17, in ros_cb: error triggers @ :

`cv_image = self.bridge.imgmsg_to_cv2(msg, "bgr8")  `

due to call in cv_bridge core.py:

`from cv_bridge.boost.cv_bridge_boost import getCvType`

**Environment information:**
 - OS version: Windows10
 - Chocolatey Versions: ros-noetic-desktop_full v20201214.0.0.2012281717

`import cv2`

... works OK.  Package is in site_packages, as well as the \cv_bridge\boost\cv_bridge_boost.lib.  But the DLL load fails.

murikill, 2021-05-25:
I had the same issue and I got it fixed. 
In fact, ros-win has already cv_bridge installed (and more other packages "on board", like [here](https://answers.ros.org/question/367248/is-possible-to-install-cv_bridge-on-ros-for-windows/) described). Therefore you don't need to compile vision_opencv in your own workspace. 
Just check with the command `rospack find cv_bridge`, if this package already exists in your ros installation directory.

ooeygui, 2021-05-27:
@murikill Thank you for the reply! 
@MikeHallettUK Does this help you, or are you experiencing another issue?

MikeHallettUK, 2021-05-28:
Thanks for the response... I have not yet had time to check it out....  

"rospack find cv_bridge" is OK and gives me = "c:\opt\ros\noetic\x64\share\cv_bridge".  I find opencv_core.dll and others in "C:\opt\ros\noetic\x64\bin" ... and will try again to get Python import to work...

Regenhardt, 2022-07-04:
I have the same problem. 

rospack find cv_bridge finds the package, same as for Mike. Python still throws the error though.

MikeHallettUK, 2022-07-04:
Sorry, I cannot remember how or if this fixed.  I gave up on Windows for a variety of frustrating reasons... and have moved all my ROS work to UBUNTU (just dual boot my PC).

MikeHallettUK, 2022-07-08:
1) I found Windows clock drifted... maybe a second per day or so... which caused synch issues with multiple ROS nodes (e.g. transforms past / future).  Tried many fixes, programmed my own 3 x nodes synch check and clock calibration reset... but in end found clock much more stable in UBUNTU.

2) ROS python tf transforms had problem due to no update of PyKDL on Windows...

3) Intel Openvino suite and Realsense were messier on Windows, and difficulties with tensorflow compatibilities:  all good on UBUNTU.

4) My on-robot nodes are UBUNTU, so making PC the same made life easier....

Mike

ooeygui, 2022-07-08:
Hi @MikeHallettUK 
Thank you very much for that enumeration.

closed 2022-07-20 by MikeHallettUK