ms-iot/ROSOnWindows #319: Unable to load planner in Melodic

kokkalisko, 2021-03-03:
**Describe the bug**

I have ported the STOMP planner for Windows and I have it working numerous PCs. In one of them, when I start any MoveIt related process and I ask for a STOMP pipeline I get the following message:

`
[0m[ERROR]** [1614765535.044611000]: Exception while loading planner 'stomp_moveit/StompPlannerManager': Failed to load library C:/Users/admin/ws/devel\bin/stomp_moveit_planner_manager.dll. Make sure that you are calling the PLUGINLIB_EXPORT_CLASS macro in the library code, and that names are consistent between this macro and your XML. Error string: Could not load library (Poco exception = C:/Users/admin/ws/devel\bin/stomp_moveit_planner_manager.dll)
Available plugins: chomp_interface/CHOMPPlanner, ompl_interface/OMPLPlanner, stomp_moveit/StompPlannerManager
`

Using the exact same `ros-melodic-desktop_full` choco distribution (i.e. 20201221.0.0.2101130001) in another computer, the error does not appear. In addition, I uninstalled and installed again the same choco distribution, but again nothing changed

I have seen a similar old issue #85 and thus, I tried to download the latest ros melodic moveit version, but it did not have any effect.  

**To Reproduce**

Unable to propose a reproduction process, since this is not specific to the STOMP planner plugin, but rather computer specific.

**Expected behavior**

Load planner plugin normally

**Environment information:**
 - OS version: `Windows 10 1809`
 - Chocolatey Versions: `ros-melodic-desktop_full 20201221.0.0.2101130001`

kokkalisko, 2021-03-06:
I also tested with other OS versions and the error does not appear. Could this be related to the OS version and an "incompatibility" with `pluginlib`?

ooeygui, 2021-03-08:
Hello @kokkalisko ,
This could be caused by a missing dependent library. The best way to diagnose this is to enable "Loader Snaps" feature in Windows. During DLL load, it will output the resolution tree - although a little cryptic, it can help identify where the missing components are.

If you could try this out:
https://ms-iot.github.io/ROSOnWindows/Porting/Cookbook.html#symbol-visibility

specifically installing the gflags tool and selecting "Show Loader snaps". I'm not clear on which .exe loads the planner, so some sleuthing would be needed to find that hosting executable. 
https://docs.microsoft.com/en-us/windows-hardware/drivers/debugger/setting-and-clearing-image-file-flags

Once you have this flag set, you can launch the node under vscode (creating a launch file which just loads that planner), and debug - you'll see the loader output in the debug Window.

If you can post the output, I can help parse it.

kokkalisko, 2021-03-16:
Hello @ooeygui,

Thank you for your valuable help! I followed your instructions and found the missing dependency. The problem was a missing vcpkg .dll, whose path needed to be added to PATH variable (along with other dlls from vcpkg).

I am closing the issue now

ooeygui, 2021-03-16:
@kokkalisko Thank you for closing the loop. The vcpkg binary should be in the path. so I'm going to reopen this issue and make sure it's handled correctly.

kokkalisko, 2021-04-15:
Unfortunately, this issue may have to reopen. 

I have the same problem with the same message but for another plugin i.e. `trac_ik` kinematics plugin. STOMP mentioned above was also a planner plugin, both of them are loaded using `pluginlib` and `poco`. 

When these plugins are built and existing in the `devel` workspace, they work fine. When they are moved to the `install` workspace or to the `opt\ROS_DISTRO\x64` folders the same message I wrote on the original post is printed .The debug window of VS code prints the following: 

`Loaded 'C:\opt\ros\melodic\x64\bin\trac_ik_kinematics_plugin.dll'. Symbols loaded.
Loaded 'C:\opt\ros\melodic\x64\bin\kdl_parser.dll'. 
Loaded 'C:\opt\ros\melodic\x64\bin\kdl_conversions.dll'. 
Unloaded 'C:\opt\ros\melodic\x64\bin\kdl_conversions.dll'.
Unloaded 'C:\opt\ros\melodic\x64\bin\kdl_parser.dll'.
Unloaded 'C:\opt\ros\melodic\x64\bin\trac_ik_kinematics_plugin.dll'.
Exception thrown at 0x00007FF8400D4B59 in move_group.exe: Microsoft C++ exception: Poco::LibraryLoadException at memory location 0x00000011166FDCD0.
Exception thrown at 0x00007FF8400D4B59 in move_group.exe: Microsoft C++ exception: class_loader::LibraryLoadException at memory location 0x00000011166FDE88.
Exception thrown at 0x00007FF8400D4B59 in move_group.exe: Microsoft C++ exception: pluginlib::LibraryLoadException at memory location 0x00000011166FE130.
[ERROR] [1618515817.841285300]: The kinematics plugin (manipulator) failed to load. Error: Failed to load library C:/opt/ros/melodic/x64\bin/trac_ik_kinematics_plugin.dll. Make sure that you are calling the PLUGINLIB_EXPORT_CLASS macro in the library code, and that names are consistent between this macro and your XML. Error string: Could not load library (Poco exception = C:/opt/ros/melodic/x64\bin/trac_ik_kinematics_plugin.dll)
[ERROR] [1618515817.841401300]: Kinematics solver could not be instantiated for joint group manipulator.`

I will try to provide more info from glags as soon as possible

ooeygui, 2021-04-15:
@kokkalisko Thank you for the report. 
I'd like to reproduce the issue locally to ensure that we unblock your work - can you share a `.repos` file so we can replicate your workspace?

kokkalisko, 2021-04-16:
I am not sure how to derive a `.repos` file is for ROS1. I just have a clean `ros-melodic-desktop_full 20201221.0.0.2101130001` installation and to further reproduce my workspace you can download [this](https://github.com/kokkalisko/trac_ik_for_windows) package and `catkin_make install -DCATKIN_BUILD_BINARY_PACKAGE=ON
`. 

Furthermore, I followed the instructions of one of your useful [tutorials](https://ms-iot.github.io/ROSOnWindows/GettingStarted/GithubSetupCD.html) for installation into `C:\opt\melodic\x64`.

I have also managed to reproduce it with the stomp planner and a another PC with both of these plugin. 

(I still owe some glags)

IoTDan, 2022-10-13:
Can you please validate if this repros on Noetic? please re-activate if it does. thanks!

closed 2022-10-13 by IoTDan