ms-iot/ROSOnWindows #323: Test & document Visibility Control Header with static built libraries

lilustga, 2021-03-17:
Test a change to the Symbol Visibility Control Header to accommodate building code as a static library. The export macros will be blank in the case of a static library.

The change is discussed in this comment thread:
https://github.com/ms-iot/ROSOnWindows/pull/322#discussion_r595002284

traversaro, 2021-03-18:
To clarify, I usually suggest to support the static usecase for standalone libraries so that they can be easily packaged inside vcpkg (https://github.com/Microsoft/vcpkg) for all the official supported triplets.

ooeygui, 2021-08-11:
https://ms-iot.github.io/ROSOnWindows/Porting/SymbolVisibility.html

ooeygui, 2021-08-11:
We need to move our symbol visibility doc to ROS.org. And add the CMake doc

IoTDan, 2022-10-13:
fixed - documented and published

closed 2022-10-13 by IoTDan