ms-iot/ROSOnWindows #324: Could you please give some help for the awesome repo of RoboStack / ros-foxy

songyuc, 2021-04-20:
Hi, guys,
I am sincerely hope you could give some help for the awesome repo of [RoboStack / ros-foxy](https://github.com/RoboStack/ros-foxy) , which is a ROS2 version based on conda.
Here are the dedicated developers, @wolfv and @Tobias-Fischer.

Appreciate it for your kind help.

ooeygui, 2021-04-20:
Thank you for the note - we are in contact with wolf about supporting and documenting the conda releases.

songyuc, 2021-04-21:
@ooeygui , apprecaite it very much!

wolfv, 2021-04-21:
I just did some slight adjustments to our build configuration and with those changes it seems that our build pipeline is progressing much more nicely: https://dev.azure.com/robostack/ros_pipelines/_build/results?buildId=1864

wolfv, 2021-04-21:
Actually we just now hit the first issue in rclcpp: https://dev.azure.com/robostack/ros_pipelines/_build/results?buildId=1864&view=logs&j=96b07928-51f6-59a2-fbfb-0ddf7ee4a6b9&t=9b369618-5551-5959-dfbc-2e0f6727780f

Maybe you have a patch for this ICE? cc @lilustga as we were recently discussing cross-platform CI using these packages.

ooeygui, 2021-04-21:
Yikes! 
```
%SRC_DIR%\ros-foxy-rclcpp\src\work\src\rclcpp\node_interfaces\node_graph.cpp(520): fatal error C1001: An internal error has occurred in the compiler.
```
That's my favorite error...

wolfv, 2021-04-21:
@traversaro suggested to move to vs2019 and I that's what we're trying now. Will give this another go tomorrow :)

ooeygui, 2021-04-21:
@wolfv oh - yea, vs2019 is really a requirement here. I've been a bit swamped on trying to put other workstreams to bed, so haven't had a chance to dig into this one.

wolfv, 2021-04-23:
alright, vs2019 does the job! https://dev.azure.com/robostack/ros_pipelines/_build/results?buildId=1886&view=logs&j=492ff20c-3425-5bf2-b87e-b67707381233&t=48b5c0cd-e63a-5512-302a-5ca53aea6517

Let's see how far we get this time :rocket: