ms-iot/ROSOnWindows #326: robot-upstart and scan-tools

Carolina710, 2021-05-19:
Dear @ooeygui , I found a project that I really would like to run on windows. In this case, it is told that exist dependencies of robot-upstart and scan-tools. I'm using melodic distribution on Win 10, and have searched for these packages on my installation and I don't found them. I founded online https://index.ros.org/p/robot_upstart/ and https://index.ros.org/p/scan_tools/

Can you confirm that these packages are compatible with Win?

Thank you very much!
Carolina Oliveira

ooeygui, 2021-05-19:
Hello @Carolina710 ,
Thank you for reaching out. I took a look at the repositories you specified. robot_upstart is a collection of python scripts specifically for manipulating the unix runtime environment. Looking at the code, these are not directly applicable to Windows.

The Scan tools do not have a melodic branch; the owner would need to port the repo melodic. Once that is completed, there are likely small changes needed to support Windows - such as exporting public symbols.

I'm curious - what is the project which depends on these libraries?

Carolina710, 2021-05-19:
Hello, thank you for replying to me. It's been a while since I used ROS, and I'm trying to catch out starting using on windows :) Thanks for your work, by the way.

I found this interesting project with wheelchairs: https://github.com/OSUrobotics/wheelchair-automation/tree/master I know it was developed for Linux and indigo distribution, but I really tried to look at each error one at a time, and robot_upstart was one of the first problems.

Thanks again :)

ooeygui, 2021-05-27:
@Carolina710 
That's a wonderful use case. Windows has some amazing accessibility tools, including things like Eye gaze tracking, which you could use for wheelchair automation. 
One trick we use to carve up a workspace in order to help bootstrap porting is to add a zero length file `CATKIN_IGNORE` in the sub-projects. This allows you to build most of the workspace. In case of robot_upstart, this is a support package, so I don't think you specifically need it. 

for the Scan_tools project, it makes use of nodelets & plugin lib. On Windows, symbols are not exported automatically. We have a doc page on this - https://ms-iot.github.io/ROSOnWindows/Porting/SymbolVisibility.html. Looking at the code, I think it would be safe to use the `set_target_properties(${LIB_NAME} PROPERTIES WINDOWS_EXPORT_ALL_SYMBOLS TRUE)` method to get the plugins to load.