ms-iot/ROSOnWindows #327: Error when using rosdep to install the dependencies

linnpl, 2021-05-31:
Hi, 

I am completely new to this and I am trying to use the plotjuggler to stream and plot from a rosnode that is running on a raspberry pie 4. I understand from the Github repository for Plotjuggler that I need to first create a catkin workspace, leading me to this webpage: https://ms-iot.github.io/ROSOnWindows/Build/source.html#configure-chocolatey-sources

So far everything went well until I got stuck on the instruction "resolving the dependencies" and the error that I got was:
`ERROR: the following packages/stacks could not have their rosdep keys resolved
to system dependencies:
perception: No definition of [image_transport_plugins] for OS [windows]
diagnostic_common_diagnostics: No definition of [python-psutil] for OS [windows]
orocos_kdl: No definition of [cppunit] for OS [windows]
rospack: No definition of [python-coverage] for OS [windows]
python_orocos_kdl: No definition of [python3-future] for OS [windows]
cv_bridge: No definition of [python-numpy] for OS [windows]
actionlib: No definition of [python-wxtools] for OS [windows]
simulators: No definition of [stage_ros] for OS [windows]
Continuing to install resolvable dependencies...
executing command [pip install -U pygraphviz]
Collecting pygraphviz
  Using cached pygraphviz-1.7.zip (118 kB)
Building wheels for collected packages: pygraphviz
  Building wheel for pygraphviz (setup.py) ... error
  ERROR: Command errored out with exit status 1:
   command: 'c:\opt\ros\noetic\x64\python.exe' -u -c 'import io, os, sys, setuptools, tokenize; sys.argv[0] = '"'"'C:\\Users\\linnl\\AppData\\Local\\Temp\\pip-install-pilr1zkp\\pygraphviz_63bd4d3f3e5b4beca4a4d99789d1b02e\\setup.py'"'"'; __file__='"'"'C:\\Users\\linnl\\AppData\\Local\\Temp\\pip-install-pilr1zkp\\pygraphviz_63bd4d3f3e5b4beca4a4d99789d1b02e\\setup.py'"'"';f = getattr(tokenize, '"'"'open'"'"', open)(__file__) if os.path.exists(__file__) else io.StringIO('"'"'from setuptools import setup; setup()'"'"');code = f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' bdist_wheel -d 'C:\Users\linnl\AppData\Local\Temp\pip-wheel-uwfmf7_k'
       cwd: C:\Users\linnl\AppData\Local\Temp\pip-install-pilr1zkp\pygraphviz_63bd4d3f3e5b4beca4a4d99789d1b02e\
  Complete output (47 lines):
  running bdist_wheel
  running build
  running build_py
  creating build
  creating build\lib.win-amd64-3.8
  creating build\lib.win-amd64-3.8\pygraphviz
  copying pygraphviz\agraph.py -> build\lib.win-amd64-3.8\pygraphviz
  copying pygraphviz\graphviz.py -> build\lib.win-amd64-3.8\pygraphviz
  copying pygraphviz\scraper.py -> build\lib.win-amd64-3.8\pygraphviz
  copying pygraphviz\testing.py -> build\lib.win-amd64-3.8\pygraphviz
  copying pygraphviz\__init__.py -> build\lib.win-amd64-3.8\pygraphviz
  creating build\lib.win-amd64-3.8\pygraphviz\tests
  copying pygraphviz\tests\test_attribute_defaults.py -> build\lib.win-amd64-3.8\pygraphviz\tests
  copying pygraphviz\tests\test_clear.py -> build\lib.win-amd64-3.8\pygraphviz\tests
  copying pygraphviz\tests\test_close.py -> build\lib.win-amd64-3.8\pygraphviz\tests
  copying pygraphviz\tests\test_drawing.py -> build\lib.win-amd64-3.8\pygraphviz\tests
  copying pygraphviz\tests\test_edge_attributes.py -> build\lib.win-amd64-3.8\pygraphviz\tests
  copying pygraphviz\tests\test_graph.py -> build\lib.win-amd64-3.8\pygraphviz\tests
  copying pygraphviz\tests\test_html.py -> build\lib.win-amd64-3.8\pygraphviz\tests
  copying pygraphviz\tests\test_layout.py -> build\lib.win-amd64-3.8\pygraphviz\tests
  copying pygraphviz\tests\test_node_attributes.py -> build\lib.win-amd64-3.8\pygraphviz\tests
  copying pygraphviz\tests\test_readwrite.py -> build\lib.win-amd64-3.8\pygraphviz\tests
  copying pygraphviz\tests\test_scraper.py -> build\lib.win-amd64-3.8\pygraphviz\tests
  copying pygraphviz\tests\test_string.py -> build\lib.win-amd64-3.8\pygraphviz\tests
  copying pygraphviz\tests\test_subgraph.py -> build\lib.win-amd64-3.8\pygraphviz\tests
  copying pygraphviz\tests\test_unicode.py -> build\lib.win-amd64-3.8\pygraphviz\tests
  copying pygraphviz\tests\__init__.py -> build\lib.win-amd64-3.8\pygraphviz\tests
  running egg_info
  writing pygraphviz.egg-info\PKG-INFO
  writing dependency_links to pygraphviz.egg-info\dependency_links.txt
  writing top-level names to pygraphviz.egg-info\top_level.txt
  reading manifest file 'pygraphviz.egg-info\SOURCES.txt'
  reading manifest template 'MANIFEST.in'
  warning: no files found matching '*.png' under directory 'doc'
  warning: no files found matching '*.txt' under directory 'doc'
  warning: no files found matching '*.css' under directory 'doc'
  warning: no previously-included files matching '*~' found anywhere in distribution
  warning: no previously-included files matching '*.pyc' found anywhere in distribution
  warning: no previously-included files matching '.svn' found anywhere in distribution
  no previously-included directories found matching 'doc\build'
  adding license file 'LICENSE'
  writing manifest file 'pygraphviz.egg-info\SOURCES.txt'
  copying pygraphviz\graphviz.i -> build\lib.win-amd64-3.8\pygraphviz
  copying pygraphviz\graphviz_wrap.c -> build\lib.win-amd64-3.8\pygraphviz
  running build_ext
[... 67 lines cut ...]
graphviz not installed. The package was not found with the source(s) listed.
 Source(s): 'https://aka.ms/ros/public'
 NOTE: When you specify explicit sources, it overrides default sources.
If the package version is a prerelease and you didn't specify `--pre`,
 the package may not be found.
Please see https://chocolatey.org/docs/troubleshooting for more
 assistance.

Chocolatey upgraded 0/1 packages. 1 packages failed.
 See the log for details (c:\opt\chocolatey\logs\chocolatey.log).

Failures
 - graphviz - graphviz not installed. The package was not found with the source(s) listed.
 Source(s): 'https://aka.ms/ros/public'
 NOTE: When you specify explicit sources, it overrides default sources.
If the package version is a prerelease and you didn't specify `--pre`,
 the package may not be found.
Please see https://chocolatey.org/docs/troubleshooting for more
 assistance.
ERROR: the following rosdeps failed to install
  pip: command [pip install -U pygraphviz] failed
  pip: Failed to detect successful installation of [pygraphviz]
  chocolatey: command [choco upgrade -y graphviz] failed
  chocolatey: Failed to detect successful installation of [graphviz]
`

linnpl, 2021-05-31:
Any comments or guidelines will be much appretiated 💯

ooeygui, 2021-06-01:
Hi there,
Thank you for reporting. Looking at the log, it seems that the root error is:
```
error: Microsoft Visual C++ 14.0 or greater is required. Get it with "Microsoft C++ Build Tools": https://visualstudio.microsoft.com/visual-cpp-build-tools/
```

Are you running these commands from a x64 Visual Studio command Window?

ooeygui, 2021-08-11:
Closing - please reopen if you are still seeing this issue.

closed 2021-08-11 by ooeygui