ms-iot/ROSOnWindows #328: Navigation2 install/setup problem - turtlebot3_fake_node build error

dfcarpenter, 2021-06-16:
**Describe the bug**
Seems to be a bug with the colcon build process building the turtlebot3_fake_node package

**To Reproduce**
Steps to reproduce the behavior:

1. Follow the steps for installing [Navigation 2 for ROSOnWindows](https://ms-iot.github.io/ROSOnWindows/ros2/nav2.html)
2. Get to the step "Build and Activate the Navigation 2 Workspace"

**Expected behavior**
I expected the colcon build process to complete without errors so I could proceed to activating the workspace
**fake_node build error log**
[stdout_stderr.log](https://github.com/ms-iot/ROSOnWindows/files/6660358/stdout_stderr.log)

**Environment information:**
 - OS version: Build 19042.985
 - Chocolatey Versions: `choco list --local-only`
```console
choco list --local-only
Chocolatey v0.10.15
chocolatey 0.10.15
chocolatey-core.extension 1.3.5.1
chocolatey-dotnetfx.extension 1.0.1
chocolatey-visualstudio.extension 1.9.0
chocolatey-windowsupdate.extension 1.0.4
dotnetfx 4.8.0.20190930
erlang 22.3
fselect 0.7.4
fzf 0.26.0
git 2.32.0
git.install 2.32.0
KB2919355 1.0.20160915
KB2919442 1.0.20160915
KB2999226 1.0.20181019
KB3033929 1.0.5
KB3035131 1.0.3
kubernetes-helm 3.5.4
python 3.9.4
python3 3.9.4
ripgrep 13.0.0
vcredist140 14.28.29325.2
vcredist2015 14.0.24215.20170201
vcxsrv 1.20.8.1
visualstudio-installer 2.0.1
visualstudio2019-workload-vctools 1.0.1
visualstudio2019buildtools 16.9.3.0
yarn 1.22.4
27 packages installed.
```

***Console output***
```console

**********************************************************************
** Visual Studio 2019 Developer Command Prompt v16.10.2
** Copyright (c) 2021 Microsoft Corporation
**********************************************************************
[vcvarsall.bat] Environment initialized for: 'x64'

C:\Windows\System32>c:\opt\ros\foxy\x64\setup.bat

C:\Windows\System32>c:\opt\ros\foxy\x64\share\gazebo\setup.bat

C:\Windows\System32>set "SDF_PATH=c:\opt\ros\foxy\x64\share\sdformat\1.6"

C:\Windows\System32>mkdir c:\nav2_ws\src

C:\Windows\System32>pushd c:\nav2_ws

c:\nav2_ws>curl -o nav2.repos https://raw.githubusercontent.com/ms-iot/ROSOnWindows/master/docs/ros2/navigation2_foxy.repos
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100   385  100   385    0     0    385      0  0:00:01 --:--:--  0:00:01  1540

c:\nav2_ws>vcs import src < nav2.repos
...
=== src\turtlebot3/turtlebot3 (git) ===
Cloning into '.'...
=== src\turtlebot3/turtlebot3_simulations (git) ===
Cloning into '.'...
=== src\utils/DynamixelSDK (git) ===
Cloning into '.'...

c:\nav2_ws>pushd c:\nav2_ws

c:\nav2_ws>colcon build
[3.350s] root DEBUG Using proactor: IocpProactor
Starting >>> dynamixel_sdk
Starting >>> turtlebot3_description
Starting >>> dynamixel_sdk_custom_interfaces
Starting >>> turtlebot3_cartographer
Starting >>> turtlebot3_example
Starting >>> turtlebot3_fake_node
Starting >>> turtlebot3_gazebo
Starting >>> turtlebot3_navigation2
Starting >>> turtlebot3_teleop
Finished <<< turtlebot3_example [4.20s]
Finished <<< turtlebot3_teleop [4.14s]
Finished <<< turtlebot3_cartographer [31.1s]
Finished <<< turtlebot3_navigation2 [31.1s]
Finished <<< turtlebot3_description [32.4s]
Finished <<< dynamixel_sdk [34.7s]
Starting >>> turtlebot3_node
Failed   <<< turtlebot3_fake_node [44.5s, exited with code 1]
Aborted  <<< turtlebot3_node [25.7s]
Aborted  <<< dynamixel_sdk_custom_interfaces [1min 47s]
Aborted  <<< turtlebot3_gazebo [2min 7s]

Summary: 6 packages finished [2min 9s]
  1 package failed: turtlebot3_fake_node
  3 packages aborted: dynamixel_sdk_custom_interfaces turtlebot3_gazebo turtlebot3_node
  1 package had stderr output: turtlebot3_gazebo
  4 packages not processed
```

ooeygui, 2021-06-17:
Thanks for the report - I'll take a look.

ooeygui, 2021-07-15:
Sorry for the delay. 
It looks like there was a change in the upstream repo for Foxy, so snapped our repos file to it:
https://github.com/ms-iot/ROSOnWindows/blob/master/docs/ros2/navigation2_foxy.repos

closed 2021-07-15 by ooeygui