ms-iot/ROSOnWindows #331: rostopic connectivity issues with Windows<->Ubuntu(VM) ROS1 (firewall?)

ipa-lth, 2021-06-23:
Hi everyone,
first of all I want to say that I am impressed how good ROS works on Windows. I have not seen such robust functioning and quality of the tutorials so far! Kudos!

I trying to connect a Windows based simulation with our ROS controller nodes. I installed ROS on Win10 an included some ROS libraries in a plugin code of the simulation. That all worked out ok.
I would like to run the simulation on windows and connect to our rosnodes running on a (local) virtualMachine with Ubuntu 18.04/melodic. I don't really care where the roscore is running, so I configured it to run on the ubuntu VM.
The communication of the nodes use normal rostopics.
Although I configured all the ROS_IP, ROS_HOSTNAME and ROS_MASTER_URI, I cannot receive data sent from the Win10 node on the Ubuntu side. It works the other way around however (receive data on the windows side, sent from ubuntu).

Suprisingly, I can see the registered node and topic (from the windows side) on the ubuntu side (rostopic list, rosnode list, ...).

I guess that some port is blocked by the windows firewall, not allowing to send data. Unfortunately, I was not able to figure out what to do. I shut down the firewall for what I am allowed to do, but the domain firewall is active, since it is handled by our IT.

Connecting to ubuntu from windows via ssh works (per IP and per name). I was not able to install ssh-server on powershell, so I could not test the other way around.

Can anyone help me? Is there a Windows Networking Tutorial somewhere?

Thanks!

gavanderhoorn, 2021-06-23:
Besides firewalls: is the VM configured to use NAT between the guest and the host?

That could also explain the `works-in-one-direction` observation you make.

ipa-lth, 2021-06-23:
No, it is configured to have NAT and also this local host-only network (which I use for ROS)

gavanderhoorn, 2021-06-23:
host-only should be ok then. It doesn't use NAT for that.

ipa-lth, 2021-06-23:
I can check later. At the moment, the ROS_HOSTNAME is set to an ip

gavanderhoorn, 2021-06-23:
You could use Wireshark to see what the nodes are trying to do. That typically very quickly shows where things go wrong.

ipa-lth, 2021-06-23:
Very good idea!

ipa-lth, 2021-06-23:
I don't know what I did, but it seems to work now. Maybe, rebooting helped?

My last action yesterday was to declare custom firewall rules for incoming and outgoing connections on port 22.

I would investigate a little more and might pose my full setup and configuration steps before closing this issue.

ooeygui, 2021-06-23:
Thank you for the comments, diagnostics and follow up! 
Thank you for the suggestion - I will do a write a tutorial on multi-compute networking, from Windows to Linux & Windows to Windows.

gavanderhoorn, 2021-07-05:
Curious: why do you set both `ROS_IP` and `ROS_HOSTNAME`? If you're going to set `ROS_HOSTNAME` to an IP address, it seems to forego the purpose of that env var.

ipa-lth, 2021-07-06:
... desperate measures...
I honestly find the envs very confusing

gavanderhoorn, 2021-07-06:
It's pretty simple really.

`ROS_IP` should be used when you don't have a fully working DNS setup available to your nodes / master.

`ROS_HOSTNAME` can be used if you do, and you gain "IP address independent" lookup (ie: IPs can change, but as long as hostnames resolve, things should still work.

If both `ROS_IP` and `ROS_HOSTNAME` are non-empty, `ROS_HOSTNAME` will be used (so when properly configured, hostnames take precedence over IPs).

While it shouldn't break anything, for a human it's confusing to have both set to IP addresses.

I would just set `ROS_IP`, and not touch `ROS_HOSTNAME` if you don't have DNS resolution working.

I would actually also recommend using an IP-based `ROS_MASTER_URI`.