ms-iot/ROSOnWindows #332: Release Galactic Chocolatey package

ooeygui, 2021-07-08:
Support the Galactic release of ROS2.

headlee, 2021-09-22:
Any ETA on when this will be released? Thanks!

ooeygui, 2021-09-22:
@headlee We're collaborating with Open Robotics and RoboStack for releasing Galactic using Condaforge - https://robostack.github.io/. I'll be updating documentation to reflect this.

ooeygui, 2021-11-15:
Closing this bug - we're recommending RoboStack moving forward.

closed 2021-11-15 by ooeygui

robogeekcanada, 2021-12-06:
Will `RoboStack` be the official way to work with ROS on Windows? 
There are so many initiatives, it's a bit confusing:

Windows11 + WSL2
VS Code with Visual Code Extension ROS

The documentation in ROS Wiki points to: https://ms-iot.github.io/ROSOnWindows/GettingStarted/Setup.html
use of Visual Studio 2019.

I'm seriously considering developing in ROS for Windows but some clarity will help of what is the path forward as the documentation is a challenge.

ooeygui, 2021-12-06:
Thank you for that feedback - I agree on clarity.

ros.org is the source of truth for ROS development. the aka.ms/ros documentation was designed to bootstrap Windows development for ROS1 and help bootstrap ROS2. It also documents our first party ROS nodes and development tooling. It does need a refresh.

RoboStack, Open Robotics and Microsoft have collaborated on making RoboStack a first class citizen for ROS2; for all platforms. It needs to be represented on the ros.org website; there is work happening for that.

Windows 11 + WSL2 is a fantastic way of developing simulation if your ultimate target is a Linux based Robot. However, WSL cannot talk directly to hardware (even with the new USB bridge), so it is limited to simulated robots.

I maintain the VSCode ROS extension; it supports ROS1, ROS2, Windows, Linux, and WSL. (I'd love to support MacOS, but don't have a Mac to test on). The VScode extension does not currently work with RoboStack, which I'll be fixing in the 0.8.0 release.

I hope this clarifies.

robogeekcanada, 2021-12-06:
Thank you @ooeygui this helps a lot. Appreciate prompt response.