ms-iot/ROSOnWindows #334: cpp nodes call abort() and produces 'Debug Error!'

cimbar, 2021-08-23:
Hi,
I cannot run c++ ros nodes on Windows. Whenever I initialize a ros node or whenever i call any function with ros namespace, the node crashes. I have tried to call these so far; `ros::init(remappings, "something")` , `ros::NodeHandle nh;` , `ros::Rate r(20)` , `ros::NodeHandle nh_private("~")` , `ros::Duration(1).sleep()`. Any of them fails and produces the error below. I didn't observe any failure on python side.

[image]

 The ROS built version is ros-noetic-desktop_full v20201214.0.0.2104301300.

Thanks,

ooeygui, 2021-08-23:
Hello There,
Are you attempting to run this code from a ROS terminal? ROS requires a specific environment, which is configured when running `c:\opt\ros\noetic\x64\setup.bat`. When this environment is not present, many dependent DLLS cannot be found, and this kind of error results.

cimbar, 2021-08-23:
Yes there is no problem on that side.

ooeygui, 2021-08-23:
Can you attach a debugger and share the stack trace?

cimbar, 2021-08-25:
The call stack below is produced when the ros::init is called. Btw I have only started roscore on ROS terminal, the node is called from VS Code but the crash is reproducable on ROS terminal too without debugging.
```
KernelBase.dll!00007ffaabd84ed9() (Unknown Source:0)
vcruntime140.dll!00007ffaa6c66480() (Unknown Source:0)
msvcp140.dll!00007ffa6781aca2() (Unknown Source:0)
roscpp.dll!00007ffa61decc70() (Unknown Source:0)
roscpp.dll!00007ffa61ded008() (Unknown Source:0)
roscpp.dll!00007ffa61f2b535() (Unknown Source:0)
roscpp.dll!00007ffa61f01315() (Unknown Source:0)
roscpp.dll!00007ffa61f01003() (Unknown Source:0)
path_creator.exe!main(int argc, char * * argv) Line 169 (c:\catkin_ws_git\src\stock_control\Warehouse_Waypoint_Creator\src\path_creator.cpp:169)
path_creator.exe!invoke_main() Line 79 (d:\a01\_work\2\s\src\vctools\crt\vcstartup\src\startup\exe_common.inl:79)
path_creator.exe!__scrt_common_main_seh() Line 288 (d:\a01\_work\2\s\src\vctools\crt\vcstartup\src\startup\exe_common.inl:288)
path_creator.exe!__scrt_common_main() Line 331 (d:\a01\_work\2\s\src\vctools\crt\vcstartup\src\startup\exe_common.inl:331)
path_creator.exe!mainCRTStartup(void * __formal) Line 17 (d:\a01\_work\2\s\src\vctools\crt\vcstartup\src\startup\exe_main.cpp:17)
kernel32.dll!00007ffaadf07034() (Unknown Source:0)
ntdll.dll!00007ffaae362651() (Unknown Source:0)
```

The exceptions:
`ros::init(argc, argv, "path_creator");`
```
CPP/std::length_error
Unhandled exception at 0x00007FFAABD84ED9 in path_creator.exe: Microsoft C++ exception: std::length_error at memory location 0x000000B8E9B7DCF0.
```

`ros::init(argc, argv, "rack_reader");`
```
CPP/std::length_error
Unhandled exception at 0x00007FFAABD84ED9 in rack_reader.exe: Microsoft C++ exception: std::length_error at memory location 0x0000002D560FCFA0.
```

`ros::NodeHandle nh;`
```
CPP/std::length_error
Unhandled exception at 0x00007FFAABD84ED9 in path_creator.exe: Microsoft C++ exception: std::length_error at memory location 0x000000A9A9EFDE00.
```

ooeygui, 2021-08-25:
That is quite interesting - it looks like your std::string might be the culprit here. is the source code available? Can you create a minimized code to share?

cimbar, 2021-08-31:
For debugging yes but I build the workspace from terminal and I got the error while I was running launch file from terminal