ms-iot/ROSOnWindows #337: collada problem with ros/gazebo

johnmichaloski, 2021-09-13:
Hello,
I am having problems displaying collada (.dae) files in gazebo using ROS URDF and Gazebo SDF that work fine in Ubuntu 18.04 and 20.04. If I use primary colors the collada works, but textures and color jpgs cause a yellow/gold crazy pallet coloring. (I'll try to attach.)

Has anyone had this problem? Windows 10  20H2 19042.1165, ROS 1, Gazebo 10.02, 

TIA
John

[image]

ooeygui, 2021-09-13:
Thanks for posting; no, we haven't had reports of this kind of issue in the past. Can you post a link to the dae? What graphics hardware are you using?

johnmichaloski, 2021-09-13:
Graphics card: 
[image]

I tested in Meshlab and the collada dae files are correct. 

[factory_shelf.zip](https://github.com/ms-iot/ROSOnWindows/files/7155948/factory_shelf.zip)

anion0278, 2023-02-09:
In the attached example the model initialization fails due to wrong mesh extension in model.sdf (.stl instead of .dae). I have fixed it, for the testing proposes: [factory_shelf_fixed.zip](https://github.com/ms-iot/ROSOnWindows/files/10697691/factory_shelf_fixed.zip)
However, this doesn't mitigate the problem with loading textures: no errors are displayed when loading the model, because gazebo seems to find the textures, but cannot load them properly (I verified this by deleting the textures - gazebo starts displaying errors about missing files). 
The issue seems to be related to #370 .