ms-iot/ROSOnWindows #343: Moveit2 run_moveit_cpp demo failing, ros2_control dies.

lilustga, 2021-11-03:
**Describe the bug**
When the run_moveit_cpp demo is run, the ros2_control node dies and the demo fails.

This seems to be due to an error with the panda_controller. Output is similar to this: https://gist.github.com/dpiet/5dc59cdddf2cc9694563dea6d27eb403

**To Reproduce**
Steps to reproduce the behavior:
 * Download the Foxy Setup which includes the 09-02 repos
 * Open terminal window for new foxy repo
 * `ros2 launch run_moveit_cpp run_moveit_cpp.launch.py`
 * Look for ros2 control process dying in the terminal.

**Environment information:**
 - OS version: 21H2

lilustga, 2021-11-03:
This was due to a version mismatch caused by some packages being migrated to `joint_state_broadcaster` but not others. With `moveit`, `moveit_resources`, `ros2_control` and `ros2_controller` all moved forward, the `run_moveit_cpp` demo plans successfully.

lilustga, 2021-11-04:
fixed by https://github.com/ms-iot/ros-windows-build/pull/206

closed 2021-11-04 by lilustga

VinHub-Gitcent, 2023-11-03:
I think this issue is again happening on the current build:

```
ros2 launch run_moveit_cpp run_moveit_cpp.launch.py
[INFO] [launch]: All log files can be found below C:\Users\ZOVNITZK\.ros\log\2023-11-03-18-54-43-995021-ZO05N005Y0-8104
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [static_transform_publisher.EXE-1]: process started with pid [16356]
[INFO] [robot_state_publisher.EXE-2]: process started with pid [33608]
[INFO] [rviz2.EXE-3]: process started with pid [33388]
[INFO] [run_moveit_cpp.EXE-4]: process started with pid [2664]
[INFO] [ros2_control_node.EXE-5]: process started with pid [3756]
[INFO] [ros2 run controller_manager spawner.py panda_arm_controller-6]: process started with pid [25108]
[INFO] [ros2 run controller_manager spawner.py panda_hand_controller-7]: process started with pid [19476]
[INFO] [ros2 run controller_manager spawner.py joint_state_broadcaster-8]: process started with pid [11332]
[static_transform_publisher.EXE-1] [INFO] [1699034084.991222300] [static_transform_publisher]: Spinning until killed publishing transform from 'world' to 'panda_link0'
[robot_state_publisher.EXE-2] Parsing robot urdf xml string.
[robot_state_publisher.EXE-2] Link panda_link1 had 1 children
[robot_state_publisher.EXE-2] Link panda_link2 had 1 children
[robot_state_publisher.EXE-2] Link panda_link3 had 1 children
[robot_state_publisher.EXE-2] Link panda_link4 had 1 children
[robot_state_publisher.EXE-2] Link panda_link5 had 1 children
[robot_state_publisher.EXE-2] Link panda_link6 had 1 children
[robot_state_publisher.EXE-2] Link panda_link7 had 1 children
[robot_state_publisher.EXE-2] Link panda_link8 had 1 children
[robot_state_publisher.EXE-2] Link panda_hand had 2 children
[robot_state_publisher.EXE-2] Link panda_leftfinger had 0 children
[robot_state_publisher.EXE-2] Link panda_rightfinger had 0 children
[robot_state_publisher.EXE-2] [INFO] [1699034085.251946800] [robot_state_publisher]: got segment panda_hand
[robot_state_publisher.EXE-2] [INFO] [1699034085.252133200] [robot_state_publisher]: got segment panda_leftfinger
[robot_state_publisher.EXE-2] [INFO] [1699034085.252156500] [robot_state_publisher]: got segment panda_link0
[robot_state_publisher.EXE-2] [INFO] [1699034085.252170500] [robot_state_publisher]: got segment panda_link1
[robot_state_publisher.EXE-2] [INFO] [1699034085.252182200] [robot_state_publisher]: got segment panda_link2
[robot_state_publisher.EXE-2] [INFO] [1699034085.252193200] [robot_state_publisher]: got segment panda_link3
[robot_state_publisher.EXE-2] [INFO] [1699034085.252203700] [robot_state_publisher]: got segment panda_link4
[robot_state_publisher.EXE-2] [INFO] [1699034085.252215000] [robot_state_publisher]: got segment panda_link5
[robot_state_publisher.EXE-2] [INFO] [1699034085.252225600] [robot_state_publisher]: got segment panda_link6
[robot_state_publisher.EXE-2] [INFO] [1699034085.252236200] [robot_state_publisher]: got segment panda_link7
[robot_state_publisher.EXE-2] [INFO] [1699034085.252246600] [robot_state_publisher]: got segment panda_link8
[robot_state_publisher.EXE-2] [INFO] [1699034085.252257400] [robot_state_publisher]: got segment panda_rightfinger
[run_moveit_cpp.EXE-4] [INFO] [1699034085.869376900] [moveit_cpp_demo]: Initialize node
[ros2_control_node.EXE-5] [INFO] [1699034086.883297100] [controller_manager]: update rate is 100 Hz
[rviz2.EXE-3] [INFO] [1699034088.206472000] [rviz2]: Stereo is NOT SUPPORTED
[rviz2.EXE-3] [INFO] [1699034088.206900900] [rviz2]: OpenGl version: 4.6 (GLSL 4.6)
[rviz2.EXE-3] [INFO] [1699034088.776208100] [rviz2]: Stereo is NOT SUPPORTED
[rviz2.EXE-3] Parsing robot urdf xml string.
[rviz2.EXE-3] [INFO] [1699034090.190592900] [moveit_rdf_loader.rdf_loader]: Loaded robot model in 0.0021705 seconds
[rviz2.EXE-3] [INFO] [1699034090.190731200] [moveit_robot_model.robot_model]: Loading robot model 'panda'...
[ros2_control_node.EXE-5] [INFO] [1699034090.954021700] [controller_manager]: Loading controller 'joint_state_broadcaster'
[ros2_control_node.EXE-5] [INFO] [1699034091.076842700] [controller_manager]: Loading controller 'panda_arm_controller'
[ros2_control_node.EXE-5] [INFO] [1699034091.219425700] [controller_manager]: Loading controller 'panda_hand_controller'
[ros2_control_node.EXE-5] [ERROR] [1699034091.219573200] [controller_manager]: Available classes:
[ros2_control_node.EXE-5] [ERROR] [1699034091.219624400] [controller_manager]: diff_drive_controller/DiffDriveController
[ros2_control_node.EXE-5] [ERROR] [1699034091.219652700] [controller_manager]: effort_controllers/JointGroupEffortController
[ros2_control_node.EXE-5] [ERROR] [1699034091.219680300] [controller_manager]: force_torque_sensor_broadcaster/ForceTorqueSensorBroadcaster
[ros2_control_node.EXE-5] [ERROR] [1699034091.219704700] [controller_manager]: forward_command_controller/ForwardCommandController
[ros2_control_node.EXE-5] [ERROR] [1699034091.219727200] [controller_manager]: imu_sensor_broadcaster/IMUSensorBroadcaster
[ros2_control_node.EXE-5] [ERROR] [1699034091.219749900] [controller_manager]: joint_state_broadcaster/JointStateBroadcaster
[ros2_control_node.EXE-5] [ERROR] [1699034091.219771900] [controller_manager]: joint_state_controller/JointStateController
[ros2_control_node.EXE-5] [ERROR] [1699034091.219798100] [controller_manager]: joint_trajectory_controller/JointTrajectoryController
[ros2_control_node.EXE-5] [ERROR] [1699034091.219820000] [controller_manager]: position_controllers/JointGroupPositionController
[... cut ...]
[ros2_control_node.EXE-5] [INFO] [1699034113.127075000] [panda_arm_controller]: Goal reached, success!
[run_moveit_cpp.EXE-4] [INFO] [1699034113.127706700] [moveit.simple_controller_manager.follow_joint_trajectory_controller_handle]: Controller panda_arm_controller successfully finished
[run_moveit_cpp.EXE-4] [INFO] [1699034113.181169100] [moveit_ros.trajectory_execution_manager]: Completed trajectory execution with status SUCCEEDED ...
[run_moveit_cpp.EXE-4] [INFO] [1699034113.181669300] [moveit.ros_planning_interface.planning_component]: Deleting PlanningComponent 'panda_arm'
```