ms-iot/ROSOnWindows #344: RViz window did not appear when run demo of Move Group C++ Interface

tangguotongling, 2021-11-10:
**Describe the bug**

Hi~

I installed  ROS melodic + MoveIt on Win10 (following https://ms-iot.github.io/ROSOnWindows/Moveit/moveit_tutorials.html)

When I run the demon of Move Group C++ Interface: 
`roslaunch panda_moveit_config demo.launch`
 (http://docs.ros.org/en/melodic/api/moveit_tutorials/html/index.html)

The RViz window did not appear, which means that I can not interact with the moveit planning progress.

However, there seemed to be a moveit.rvz task on the task bar:
[image]

 and I got the tail of console output as :

`QObject::connect: Cannot queue arguments of type 'QVector<int>'`
`(Make sure 'QVector<int>' is registered using qRegisterMetaType().)`
`QObject::connect: Cannot queue arguments of type 'QVector<int>'`
`(Make sure 'QVector<int>' is registered using qRegisterMetaType().)`

but I do not think that relates with the phenomena.

What's more, I can run the demo of MoveIt Quickstart in RViz correctly:
`roslaunch panda_moveit_config demo.launch rviz_tutorial:=true`
(http://docs.ros.org/en/melodic/api/moveit_tutorials/html/doc/quickstart_in_rviz/quickstart_in_rviz_tutorial.html)

Can anybody help me?

Thank you~

**To Reproduce**
Steps to reproduce the behavior:
* Catkin workspace setup instructions

`call c:\opt\ros\melodic\x64\setup.bat
`
  * Clone locations
```
mkdir c:\moveit_ws\src
cd c:\moveit_ws\src
git clone https://github.com/ros-planning/moveit_tutorials.git -b melodic-devel
git clone https://github.com/ros-planning/panda_moveit_config.git -b melodic-devel
git clone https://github.com/frankaemika/franka_ros-release -b release/kinetic/franka_description
```
  * build command line
```
cd c:\moveit_ws
catkin_make
```
  * Launch command line
`c:\moveit_ws>call devel\setup.bat`
`roslaunch panda_moveit_config demo.launch`

**Expected behavior**

According to the moveit, there ought to be a RViz window with a robot like:
[image]
where I can interact with the MoveIt.
  
**Environment information:**
 - OS version: `winver` 
      Win10 Pro 21H1 19043.1320
 - Chocolatey Versions: `choco list --local-only`
      Chocolatey v0.11.3
      chocolatey 0.11.3
      chocolatey-core.extension 1.3.5.1
      git 2.33.1
      git.install 2.33.1
[... 270 lines cut ...]
*     - ExecuteTrajectoryAction
*     - GetPlanningSceneService
*     - KinematicsService
*     - MoveAction
*     - PickPlaceAction
*     - MotionPlanService
*     - QueryPlannersService
*     - StateValidationService
********************************************************

[ INFO] [1636530112.387739300]: MoveGroup context using planning plugin ompl_interface/OMPLPlanner
[ INFO] [1636530112.388510000]: MoveGroup context initialization complete
[92m
You can start planning now!

[0m[ INFO] [1636530113.423653000]: Stereo is NOT SUPPORTED
[ INFO] [1636530113.423827300]: OpenGl version: 4.6 (GLSL 4.6).
[ INFO] [1636530113.533963000]: Loading robot model 'panda'...
[ INFO] [1636530113.922282900]: Loading robot model 'panda'...
[ INFO] [1636530114.060339000]: Starting planning scene monitor
[ INFO] [1636530114.062917100]: Listening to '/move_group/monitored_planning_scene'
QObject::connect: Cannot queue arguments of type 'QVector<int>'
(Make sure 'QVector<int>' is registered using qRegisterMetaType().)
QObject::connect: Cannot queue arguments of type 'QVector<int>'
(Make sure 'QVector<int>' is registered using qRegisterMetaType().)

lilustga, 2021-11-12:
Hello @tangguotongling,

I tried running the demo on the latest tutorial and was able to see this same behavior by omitting `rviz_tutorial:=true` from the command. Could you try running ` roslaunch panda_moveit_config demo.launch rviz_tutorial:=true`. That is the command as recommended in the quickstart tutorial. http://docs.ros.org/en/melodic/api/moveit_tutorials/html/doc/quickstart_in_rviz/quickstart_in_rviz_tutorial.html

Please let me know if that works. Thanks!

tangguotongling, 2021-11-15:
> Hello @tangguotongling,
>
> I tried running the demo on the latest tutorial and was able to see this same behavior by omitting `rviz_tutorial:=true` from the command. Could you try running ` roslaunch [...]

Hi~

Thank you for your attention~
I tried the "MoveIt Quickstart in RViz"(http://docs.ros.org/en/melodic/api/moveit_tutorials/html/doc/quickstart_in_rviz/quickstart_in_rviz_tutorial.html) command as you required:
` roslaunch panda_moveit_config demo.launch rviz_tutorial:=true`
Yes, I can see the RViz window. 
But if I continue with the "Move Group C++ Interface":(http://docs.ros.org/en/melodic/api/moveit_tutorials/html/doc/move_group_interface/move_group_interface_tutorial.html)
`roslaunch panda_moveit_config demo.launch`
`roslaunch moveit_tutorials move_group_interface_tutorial.launch`
RViz did not show which prevent the interactive with the demo since it requires a click on the 'Next' button.
I can shield the visualization related lines in the "Move Group C++ Interface".
But that degenerates to a pure console window.......

tangguotongling, 2021-11-16:
@tangguotongling see this