ms-iot/ROSOnWindows #347: RQT Image View Plugin Failed to Load in Noetic (Windows 10)

shivarajd, 2021-11-14:
**Describe the bug**

I get this error when I try to launch rqt_image_view in Noetic running on Windows 10.

> [ERROR] [1636899511.027524100]: Failed to load nodelet [rqt_image_view/ImageView_0] of type [rqt_image_view/ImageView]: Failed to load library c:\opt\ros\noetic\x64\bin/rqt_image_view.dll. Make sure that you are calling the PLUGINLIB_EXPORT_CLASS macro in the library code, and that names are consistent between this macro and your XML. Error string: Could not load library (Poco exception = c:\opt\ros\noetic\x64\bin/rqt_image_view.dll)
RosPluginlibPluginProvider::load_explicit_type(rqt_image_view/ImageView) failed creating instance
PluginManager._load_plugin() could not load plugin "rqt_image_view/ImageView": RosPluginlibPluginProvider.load() could not load plugin "rqt_image_view/ImageView"

**To Reproduce**
Steps to reproduce the behavior:
* Run `roscore` on ROS command prompt
* Run `rqt_image_view` on another ROS command prompt

**Expected behavior**
Launch rqt_image_view

**Environment information:**
 - OS version: Windows 10 Version 21H1 (19043.1288)
 - Chocolatey Versions: 
Chocolatey v0.11.2
chocolatey-core.extension 1.3.5.1
chocolatey-windowsupdate.extension 1.0.4
KB2919355 1.0.20160915
KB2919442 1.0.20160915
KB2999226 1.0.20181019
KB3033929 1.0.5
KB3035131 1.0.3
ros-foxy-desktop 20201211.0.0.2104300001
ros-noetic-desktop_full 20201214.0.0.2104301300
vcredist140 14.29.30135
vcredist2010 10.0.40219.32503

ooeygui, 2021-11-15:
Hi @shivarajd,
It looks like we have this corrected in the noetic prerelease which we are preparing for launch. I need to bump the version. We've been selfhosting it for a bit, and looking good; I had hoped to get one more rviz fix in before release.

Can you upgrade to `choco upgrade ros-noetic-desktop_full --pre`, then try your scenarios?

ooeygui, 2021-11-15:
Thank you for trying the latest pre-release. I need to dig into this a little deeper; your reddit post shows that rqt_image_view is loading libpng; my version launches without loading libpng. 

May I connect with you for a quick screen share? I sent a connection request through linkedin.