ms-iot/ROSOnWindows #348: Noetic-MoveIt catkin_make failed

tangguotongling, 2021-11-15:
Hi, it's me again.
To fix my previous issue, I cleaned the moveit workspace and clone/build it.
But this time I did not pass the catkin_make process
Here is the phenomena:

**Describe the bug**

Catkin_make MoveIt failed for Noetic on Windows 10

**To Reproduce**

I followed the article "MoveIt Tutorials on Windows"(https://ms-iot.github.io/ROSOnWindows/Moveit/moveit_tutorials.html)

* Catkin workspace setup instructions 
`mkdir c:\opt\chocolatey`
`set ChocolateyInstall=c:\opt\chocolatey`
`choco source add -n=ros-win -s="https://aka.ms/ros/public" --priority=1`
`choco upgrade ros-noetic-desktop_full -y --execution-timeout=0`

* Clone locations
`mkdir c:\moveit_ws\src`
`cd c:\moveit_ws\src`

`git clone https://github.com/ros-planning/moveit_tutorials.git -b master`
`git clone https://github.com/ros-planning/panda_moveit_config.git -b melodic-devel`
`git clone https://github.com/frankaemika/franka_ros-release -b release/kinetic/franka_description`

* build command line
`cd c:\moveit_ws`
`catkin_make`

**Expected behavior**

However, the catkin_make process failed which reported errors like: 
`C:\moveit_ws\src\moveit_tutorials\doc\moveit_cpp\src\moveit_cpp_tutorial.cpp(34): error C2653: 'moveit_cpp': is not a class or namespace name
`
`...`
`ninja: build stopped: subcommand failed.`
`Invoking "ninja -j8 -l8" failed`

Has anyone met the same problem on Noetic? 
Could you help me to fix it out? 
I quote the main error section at the bottom and attached the whole console output at the end (about 2000 lines......)
And I will try the melodic version to see if it will happen.

**Environment information:**
 - OS version: `winver`
Windows10 21H1  19043.1320
 - Chocolatey Versions: `choco list --local-only`
Chocolatey v0.11.3
chocolatey-core.extension 1.3.5.1
chocolatey-windowsupdate.extension 1.0.4
KB2919355 1.0.20160915
KB2919442 1.0.20160915
KB2999226 1.0.20181019
KB3033929 1.0.3
KB3035131 1.0.3
ros-foxy-desktop 20201211.0.0.2104300001
ros-melodic-desktop_full 20201221.0.0.2104300002
ros-noetic-desktop_full 20201214.0.0.2104301300
vcredist140 14.30.30704
vcredist2010 10.0.40219.32503
12 packages installed.

**main error output**
-------------------------
[36/54] Building CXX object moveit_tutorials\doc\moveit_cpp\CMakeFiles\moveit_cpp_tutorial.dir\src\moveit_cpp_tutorial.cpp.obj
FAILED: moveit_tutorials/doc/moveit_cpp/CMakeFiles/moveit_cpp_tutorial.dir/src/moveit_cpp_tutorial.cpp.obj
C:\PROGRA~2\MICROS~3\2019\ENTERP~1\VC\Tools\MSVC\1428~1.299\bin\Hostx64\x64\cl.exe  /nologo /TP -DNOMINMAX -DROSCONSOLE_BACKEND_LOG4CXX -DROS_BUILD_SHARED_LIBS=1 -DROS_PACKAGE_NAME=\"moveit_tutorials\" -DWIN32_LEAN_AND_MEAN -D_USE_MATH_DEFINES -IC:\moveit_ws\src\moveit_tutorials\doc\interactivity\include -IC:\opt\ros\noetic\x64\include -IC:\opt\ros\noetic\x64\share\xmlrpcpp\cmake\..\..\..\include\xmlrpcpp -IC:\opt\ros\noetic\x64\include\eigen3 -IC:\opt\ros\noetic\x64\share\urdfdom\..\..\include -IC:\opt\ros\noetic\x64\share\urdfdom_headers\..\..\include -IC:\opt\ros\noetic\x64\include\bullet -IC:\opt\ros\noetic\x64\share\orocos_kdl\..\..\include -IC:\opt\ros\noetic\x64\share\orocos_kdl\..\..\include\eigen3 /DWIN32 /D_WINDOWS /W3 /GR /EHsc /MD /Zi /O2 /Ob1 /DNDEBUG   /D _VARIADIC_MAX=10 /Zc:__cplusplus -std:c++14 /showIncludes /Fomoveit_tutorials\doc\moveit_cpp\CMakeFiles\moveit_cpp_tutorial.dir\src\moveit_cpp_tutorial.cpp.obj /Fdmoveit_tutorials\doc\moveit_cpp\CMakeFiles\moveit_cpp_tutorial.dir\ /FS -c C:\moveit_ws\src\moveit_tutorials\doc\moveit_cpp\src\moveit_cpp_tutorial.cpp
C:\opt\ros\noetic\x64\include\boost/bind.hpp(41): note: The practice of declaring the Bind placeholders (_1, _2, ...) in the global namespace is deprecated. Please use <boost/bind/bind.hpp> + using namespace boost::placeholders, or define BOOST_BIND_GLOBAL_PLACEHOLDERS to retain the current behavior.
[... 14 lines cut ...]
C:\moveit_ws\src\moveit_tutorials\doc\moveit_cpp\src\moveit_cpp_tutorial.cpp(86): error C3536: 'plan_solution1': cannot be used before it is initialized
C:\moveit_ws\src\moveit_tutorials\doc\moveit_cpp\src\moveit_cpp_tutorial.cpp(89): error C3536: 'robot_start_state': cannot be used before it is initialized
C:\moveit_ws\src\moveit_tutorials\doc\moveit_cpp\src\moveit_cpp_tutorial.cpp(95): error C3536: 'joint_model_group_ptr': cannot be used before it is initialized
C:\moveit_ws\src\moveit_tutorials\doc\moveit_cpp\src\moveit_cpp_tutorial.cpp(125): error C3536: 'start_state': cannot be used before it is initialized
C:\moveit_ws\src\moveit_tutorials\doc\moveit_cpp\src\moveit_cpp_tutorial.cpp(131): error C3536: 'plan_solution2': cannot be used before it is initialized
C:\moveit_ws\src\moveit_tutorials\doc\moveit_cpp\src\moveit_cpp_tutorial.cpp(133): error C2664: 'moveit::core::RobotState::RobotState(const moveit::core::RobotModelConstPtr &)': cannot convert argument 1 from 'int' to 'const moveit::core::RobotModelConstPtr &'
C:\moveit_ws\src\moveit_tutorials\doc\moveit_cpp\src\moveit_cpp_tutorial.cpp(133): note: Reason: cannot convert from 'int' to 'const moveit::core::RobotModelConstPtr'
C:\moveit_ws\src\moveit_tutorials\doc\moveit_cpp\src\moveit_cpp_tutorial.cpp(133): note: No constructor could take the source type, or constructor overload resolution was ambiguous
C:\opt\ros\noetic\x64\include\moveit/robot_state/robot_state.h(94): note: see declaration of 'moveit::core::RobotState::RobotState'
C:\moveit_ws\src\moveit_tutorials\doc\moveit_cpp\src\moveit_cpp_tutorial.cpp(163): error C2100: illegal indirection
C:\moveit_ws\src\moveit_tutorials\doc\moveit_cpp\src\moveit_cpp_tutorial.cpp(176): error C3536: 'plan_solution3': cannot be used before it is initialized
C:\moveit_ws\src\moveit_tutorials\doc\moveit_cpp\src\moveit_cpp_tutorial.cpp(178): error C2664: 'moveit::core::RobotState::RobotState(const moveit::core::RobotModelConstPtr &)': cannot convert argument 1 from 'int' to 'const moveit::core::RobotModelConstPtr &'
C:\moveit_ws\src\moveit_tutorials\doc\moveit_cpp\src\moveit_cpp_tutorial.cpp(178): note: Reason: cannot convert from 'int' to 'const moveit::core::RobotModelConstPtr'
C:\moveit_ws\src\moveit_tutorials\doc\moveit_cpp\src\moveit_cpp_tutorial.cpp(178): note: No constructor could take the source type, or constructor overload resolution was ambiguous
C:\opt\ros\noetic\x64\include\moveit/robot_state/robot_state.h(94): note: see declaration of 'moveit::core::RobotState::RobotState'
C:\moveit_ws\src\moveit_tutorials\doc\moveit_cpp\src\moveit_cpp_tutorial.cpp(219): error C3536: 'plan_solution4': cannot be used before it is initialized
C:\moveit_ws\src\moveit_tutorials\doc\moveit_cpp\src\moveit_cpp_tutorial.cpp(221): error C2664: 'moveit::core::RobotState::RobotState(const moveit::core::RobotModelConstPtr &)': cannot convert argument 1 from 'int' to 'const moveit::core::RobotModelConstPtr &'
......
[43/54] Building CXX object moveit_tutorials\doc\subframes\CMakeFiles\subframes_tutorial.dir\src\subframes_tutorial.cpp.obj
C:\opt\ros\noetic\x64\include\boost/bind.hpp(41): note: The practice of declaring the Bind placeholders (_1, _2, ...) in the global namespace is deprecated. Please use <boost/bind/bind.hpp> + using namespace boost::placeholders, or define BOOST_BIND_GLOBAL_PLACEHOLDERS to retain the current behavior.
C:\moveit_ws\src\moveit_tutorials\doc\subframes\src\subframes_tutorial.cpp(215): warning C4267: 'initializing': conversion from 'size_t' to 'int', possible loss of data
ninja: build stopped: subcommand failed.
Invoking "ninja -j8 -l8" failed

[ms-noetic-moveit-catkinmake.log](https://github.com/ms-iot/ROSOnWindows/files/7535584/ms-noetic-moveit-catkinmake.log)

lilustga, 2021-11-15:
Hi @tangguotongling,

This is a known breaking change between the Noetic tutorials and the current Noetic on Windows release for moveit. We have work planned to fix it.

Did the Melodic tutorials work for you in the previous issue you opened?

tangguotongling, 2021-11-16:
Hi @lilustga 

Thank you for your attention~

Yes, I turned back to ROS-Melodic again.

As you suggested I tried the command in "MoveIt Quickstart in RViz"(http://docs.ros.org/en/melodic/api/moveit_tutorials/html/doc/quickstart_in_rviz/quickstart_in_rviz_tutorial.html) 
`roslaunch panda_moveit_config demo.launch rviz_tutorial:=true`
Yes, I can see the RViz window this time.
But if I continue with the "Move Group C++ Interface":(http://docs.ros.org/en/melodic/api/moveit_tutorials/html/doc/move_group_interface/move_group_interface_tutorial.html)
`roslaunch panda_moveit_config demo.launch`
`roslaunch moveit_tutorials move_group_interface_tutorial.launch`
RViz did not appear, which prevent the interactive with the demo since it requires a click on the 'Next' button.
I can shield the RViz visualization related lines in the source code of "Move Group C++ Interface".
But that degenerates to a pure console window.......
Any more suggestion?  O(∩_∩)O

tangguotongling, 2021-11-16:
@tangguotongling test now

FeiyuYan95, 2021-11-16:
@tangguotongling  hi

lilustga, 2021-11-17:
Hi @tangguotongling,

Could you try deleting your moveit workspace and going through the instructions again now?