ms-iot/ROSOnWindows #356: Visual Studio Community 2019 no longer available.

KeithGal, 2021-11-19:
All Visual Studio Community downloads point to the 2022 version, which is not supported by ROS yet.

I have just setup a new PC and can't build anything.

  File "c:\opt\ros\foxy\x64\lib\site-packages\colcon_cmake\task\cmake\build.py", line 162, in _reconfigure
    raise RuntimeError(
RuntimeError: Unknown / unsupported VS version '17.0'

KeithGal, 2021-11-20:
Partial answer:

If CMake is updated to the latest version, then newly created packages can be built as they default to VS 2022 and the latest CMake version supports this.

However old packages still can't be rebuilt.

ooeygui, 2021-11-23:
Visual Studio 2019 is available in the older downloads section - https://visualstudio.microsoft.com/vs/older-downloads/. I'll update the docs.

ooeygui, 2021-11-23:
The default toolchains are the important part.

ooeygui, 2021-11-23:
I updated ros.org for ROS1 and fixed out ROS2 docs - https://github.com/ms-iot/ROSOnWindows/pull/357.

KeithGal, 2021-11-25:
Thankyou, I did look but obviously not hard enough.

IoTDan, 2022-10-10:
Per @ooeygui comments above

closed 2022-10-10 by IoTDan