ms-iot/ROSOnWindows #363: Gazebo isn't working in windows 10 with ros2 after following the tutorial

marpeja, 2022-03-18:
I have followed the tutorial of installing ros2 in windows 10 and completed it succesfully.

However, gazebo is still not recognized by my computer:

[image]

It says that gazebo is not recognized as an iternal or external command.

I continued with the instalation of Nav2 as explained in your tutorial to see if this was normal, because as I said, the installation was succesful.

Moreover, when building the nav2_ws workspace, this is what I get:

[image]

So I guess Gazebo isn't installed correctly...

Any suggestions?

ooeygui, 2022-03-18:
Thank you for the report.
The ignition repository doesn't supply a `gazebo` launch for Windows. However, it can be launched via launch file. The nav2 documentation describes how to launch it.

The build warning you are seeing is about cmake not liking the package name mismatch. This is a warning and doesn't affect launch.

please do note that launching gazebo can take several minutes the first time without any feedback as it is downloading files from the internet using it's UI thread.

marpeja, 2022-03-21:
@ooeygui Thank you so much for your response! 

I finally think I make it work, however Gazebo has been launching for more than hour now and it hasn't finished yet...

Yesterday I closed it after 30 minutes and I tried to launch it today but it didn't work, so I erased ROS2 and NAV2 and installed them again. Now it launches, but as I said, it's been more than an hour since it has been launched without finishing.

What can I do?