ms-iot/ROSOnWindows #365: Robot not moving when setting navigation2 goal in Rviz

marpeja, 2022-03-30:
Hey!

After launching Gazebo and Rviz, I set the "2D Pose Estimate" and I am able to see the robot and particle cloud.

However, when setting the Navigation2 Goal, it appears a track but the robot doesn't follow it.

I looked in the cmd and saw that Rviz is returning the following information:

`[rviz2.EXE-10] [INFO] [1648657815.326225200] [rviz2]: Message Filter dropping message: frame 'base_scan' at time 485.407 for reason 'Unknown'`
`[rviz2.EXE-10] [INFO] [1648657815.328628600] [rviz2]: Message Filter dropping message: frame 'odom' at time 1648657811.066 for reason 'Unknown'`
`[rviz2.EXE-10] [ERROR] [1648657815.421274400] [rviz2]: Error prior to using GLSL Program Object : GL_INVALID_OPERATION`
`[rviz2.EXE-10] Validation successful.`

Any idea of what can I do?

Thank you in advance!

ooeygui, 2022-03-30:
Thank you for reaching out. 
Can you tell me what version of ROS2 you have installed? 
Did you install using the chocolatey package?

The GL_INVALID_OPERATION is concerning - what kind of GPU are you using?

marpeja, 2022-03-30:
ROS2 ros-foxy-desktop, the one shown in the tutorial, which I followed, so yes I installed it using chocolatey.

And my GPU is the Nvidia RTX2060.