ms-iot/ROSOnWindows #370: Gazebo textures aren't loading

ooeygui, 2022-04-22:


LuisYordano, 2022-05-18:
Hi Lou,

I am testing **ROS Noetic on Windows 10**, the textures and models aren't load correctly in **Gazebo 10**.

I am attaching some screenshots :
---------------------------------------
**Robot Fetch**

[image]

[image]

**Turtlebot3:**

[image]

[image]

-------------------------------------

Do you have any workaround for this issue?

Thank you in advance.

anion0278, 2023-02-09:
@LuisYordano I faced a similar/related problem in ROS2 Gazebo 10.2.0. It seems that the issue with ...chdir to _'C:\Users\<user>\.gazebo/models'_ can be solved by manually creating _'models'_ dir in _'C:\Users\<user>\.gazebo'_ . This also solves problems of loading models from Gazebo online database (from the Insert tab -> http://models.gazebosim.org/). 
Some models, such as "Bookshelf" are correctly loaded including their textures; this fix however does not solve the problem of texture loading for most of the models (Bus, SUV, PR2, etc.), even though their textures are available in _'C:\Users\<user>\.gazebo/models/<model_name>/materials'_ . 

[image]

Additionally, loading models such as "Kuka YouBot" crashes gazebo:
```
[gzclient --verbose   -2] [Err] [C:\a\1\s\ws\src\gazebo\gazebo10\gazebo\rendering\RenderEngine.cc:544] EXCEPTION: Unable to load Ogre Resources.
[gzclient --verbose   -2] Make sure theresources path in the world file is set correctly.
[gzclient --verbose   -2]
[ERROR] [gzclient --verbose   -2]: process has died [pid 14136, exit code 3221225477, cmd 'gzclient --verbose   '].
```
I guess, the _'Models'_ dir should have been created automatically during first run of Gazebo (this is the time when _.gazebo_ dir is initialized). Tested it on two Win10 machines.