ms-iot/ROSOnWindows #371: ModuleNotFoundError: No module named 'cv_bridge.boost.cv_bridge_boost' while using opencv in ROS2

sssxrf, 2022-05-28:
I'm following the instructions in [https://automaticaddison.com/getting-started-with-opencv-in-ros-2-foxy-fitzroy-python/](url) to create a small pub/sub to get image from my camera. Then I encountered a problem when I ran the img_subscriber. 
[image]
However, the publisher works just fine.
[image]
Please help me solve this!

ooeygui, 2022-05-30:
Tagging this as well https://github.com/ros-perception/vision_opencv/issues/247

ooeygui, 2022-05-30:
and more https://github.com/ms-iot/ROSOnWindows/issues/317

ooeygui, 2022-05-30:
Thank you for the repro case, I'll take a look this week

francocipollone, 2022-09-07:
> Thank you for your response. I find cv_bridge with `ros2 pkg prefix cv_bridge`. But still it throws the error.

Hi there, were you able to workaround this issue?

ooeygui, 2022-10-03:
Working on testing out a fix for this. Stay tuned.

ooeygui, 2022-10-06:
I have a build in testing which resolves this. If you want a pre-preview, you can download it from our continuous integration server:
https://ros-win.visualstudio.com/bed058dd-46da-4029-bb85-25eae7674d09/_apis/build/builds/15979/artifacts?artifactName=ros-foxy-setup-signed&api-version=7.0&%24format=zip

Uranophan, 2023-01-18:
> I have a build in testing which resolves this. If you want a pre-preview, you can download it from our continuous integration server: [...]

Is there any update on how to fix the problem? Your link seems to be outdated. When trying to execute my code I get the error No module named 'cv_bridge.boost.cv_bridge_boost'. Open cv is installed as per the installation.

jiawei-gao056, 2023-01-26:
I had the same problem when using humble on win10. No module named 'cv_bridge.boost.cv_bridge_boost' shows when I run the img_subscriber. How to solve please?

sssxrf, 2023-01-26:
Sorry for the late reply. I didn't use cv_bridge but add the dependencies locally by myself.

jiawei-gao056, 2023-01-26:
I am new to ROS and don't know how to do that. Could you share how to add the dependencies manually?

ooeygui, 2023-01-26:
@jiawei-gao056 Which instructions did you use to install ROS?

ooeygui, 2023-01-26:
Reopening on humble

jiawei-gao056, 2023-01-26:
This one. https://docs.ros.org/en/humble/Installation/Windows-Install-Binary.html.

jiawei-gao056, 2023-01-26:
This one. https://docs.ros.org/en/humble/Installation/Windows-Install-Binary.html

zsedrovitstamas, 2023-02-23:
I have the same issue. I have tried the binary installation both from [MS](https://ms-iot.github.io/ROSOnWindows/GettingStarted/SetupRos2.html) and [ROS](https://docs.ros.org/en/foxy/Installation/Windows-Install-Binary.html) I have built [ROS Foxy from source](https://docs.ros.org/en/foxy/Installation/Alternatives/Windows-Development-Setup.html) on Win11, and vision_opencv with boost, but no luck. Is there any solution yet?

mysunshine01, 2023-09-20:
[image]

any solution yet? I'm facing the same issue here.

ooeygui, 2023-09-20:
@mysunshine01 The python version included with noetic does not include the full python release and SDK. This is the root cause of this error. Microsoft is no longer releasing ROS1 builds, so this will not be fixed in-distro.

A potential solutions:
* Use https://robostack.github.io/ for ROS1 builds for Windows. 
* Attempt to overlay the full Python build and SDK into distro, replacing the distribution in noetic.