ms-iot/ROSOnWindows #377: cartographer/cartographer_ros 2.0 build for ROS2 Humble

AustinMooreResonantSciences, 2022-08-04:
Hello, we are trying to build cartographer + cartographer_ros 2.0 on Windows 10 ROS 2 Humble (binary install) with MSVC (have also tried Clang). After a lot of effort we can get it to build, but we are running into a segfault once the cartographer node starts getting LiDAR PointCloud2 messages. More detail [here](https://github.com/ros2/cartographer_ros/issues/64).

We noticed that there is a humble branch on the [ROS Windows Build repo](https://github.com/ms-iot/ros-windows-build/tree/feature/humble) with a commit that actually removed Cartographer. Couple of questions:

1. Will this repo auto-build Humble for Windows and have it available on a package manager (similar to ros2-foxy-desktop on Chocolatey)?
2. Has anyone here attempted building Cartographer 2.0 on Windows and had success running? Was it removed because it wasn't building/running?

If there is a better place to ask this please let me know! Thanks for any help.

ooeygui, 2022-08-05:
@AustinMooreResonantSciences Hi there,
I would absolutely love to know more about your work with ROS on Windows. Could you connect with me on linked in? https://www.linkedin.com/in/louamadio/

We are in the process of bringing up a Microsoft supported ROS2 build for Humble. We are migrating to vs2022 and partitioning the build space to efficiently bring it up. Removing cartographer from the build was just expediency, it will be added back when we do the nav2 bringup.

We will have ros-humble-desktop available soon, as well as gazebo 11. We're actively working on it.

AustinMooreResonantSciences, 2022-08-05:
Thanks for the quick response!

Will the build you bring up for humble be available on a package manager similar to how the foxy release was on Chocolatey?

If we get Cartographer working and you have any questions about what we did to make it work when you start working on your nav2 bringup feel free to ping me!

ooeygui, 2022-08-05:
Yes, it will be available as `ros-humble-desktop` on chocolatey, and I'll update docs on ros.org and http://aka.ms/ros. 

Thank you for the offer on cartographer! I'll definitely reach out if we have issues.

And thank you for your interest in ROS2 on Windows!

headlee, 2022-08-30:
I tried to build the 1.0 version of cartographer+cartographer_ros with Humble (using vcpkg for cartographer and https://github.com/ms-iot/cartographer_ros/tree/1.0.9001/windows for cartographer_ros) and MSVC 2019. Had a few build issues but was able to get through them and am now getting the same crash described above. I did have to bump `CMAKE_CXX_STANDARD` to 17 to get it to build and got a bunch of warnings...

As far as I can tell the versions I used are the same that would have been used for the Chocolatey foxy build which works fine. I've also been able to run everything fine in WSL2 Ubuntu so I'm wondering if there is a MSVC-specific issue here? I think I'll try building with VS2022 next.

Looking forward to seeing an official package for Humble!

ooeygui, 2022-08-30:
@headlee Thanks for your interest! We are definitely getting close. You can follow along with the workstream by monitoring our build server - https://ros-win.visualstudio.com/ros-win/_build?definitionId=150. Technically you can download setup.exe and run it today. I'm currently working on Gazebo 11, so we can bring up Nav2 and MoveIt2.

headlee, 2023-04-04:
> @headlee This is actively being worked on and tested. With the holidays approaching, our velocity will be slowing.

Hi @ooeygui, just wanted to follow up again and see if there was any progress on getting Cartographer+Cartographer-ROS into the Humble installer. Thanks!