ms-iot/ROSOnWindows #378: time_parameterization.h missing

qualia0000, 2022-08-23:
I am facing a problem when installing Deep_grasp_demo package in ROS melodic in Windows. The error is as following:
`moveit_task_constructor\core\include\moveit/task_constructor/merge.h(42): fatal error C1083: Cannot open include file: 'moveit/trajectory_processing/time_parameterization.h': No such file or directory`

I searched the folder trajectory_processsing finding that no file named time_parameterization.h. In MoveIt git repo, the file is included in the folder. Besides, in MoveIt git repo, the header file iterative_time_parameterization.h has included time_parameterization.h but in the Windows version it does not. 

**Environment information:**
 - OS version: Windows 10 21H2
 - Chocolatey Versions: Chocolatey v1.1.0
 
Appreciated for every help.

ooeygui, 2022-08-23:
Thank you for the report.

## Investigation
We snapped the melodic release of MoveIt in https://github.com/ms-iot/ros-windows-build/blob/master/ros/melodic/melodic_override.repos with some minor fixes for windows (which were later up streamed):

```yaml
  moveit:
    type: git
    url: https://github.com/ms-iot/moveit.git
    version: windows/1.0.7
```
This was snapped 2 years ago - the last time we touched Melodic, and not moved forward.

It looks like the Time parameterization header was added on January 6th 2022:
https://github.com/ros-planning/moveit/commits/3ba0a077a219d8ecea5465e19a886ce28faa6e36/moveit_core/trajectory_processing/include/moveit/trajectory_processing/time_parameterization.h

There are some options here:
* You can include the current tip of tree of moveit in your project repo. It should override the MoveIt build in `c:\opt`
* We would need to respin melodic with an updated moveit. 

Can you tell me more about what you are trying to do? Is this for a commercial project?

qualia0000, 2022-08-24:
@ooeygui Thank you for your reply. Would you mind elaborating the including steps? Should I clone the latest version of moveit-melodic repo from [moveit](https://github.com/ros-planning/moveit)? I am afraid without Chocolatey the moveit cannot be successfully compiled.

qualia0000, 2022-08-24:
Update:
I am trying the method in [https://moveit.ros.org/install/source-windows/](https://moveit.ros.org/install/source-windows/) to build MoveIt repo from source. However the wstool is not working on ROS Windows. I also used `vcs` command as alternative but there is an error when executing it:
`File "C:\opt\ros\melodic\x64\Scripts\vcs-script.py", line 11, in <module>
    load_entry_point('vcstool==0.3.0', 'console_scripts', 'vcs')()
  File "C:\opt\ros\melodic\x64\lib\site-packages\pkg_resources\__init__.py", line 489, in load_entry_point
    return get_distribution(dist).load_entry_point(group, name)
  File "C:\opt\ros\melodic\x64\lib\site-packages\pkg_resources\__init__.py", line 2852, in load_entry_point
    return ep.load()
  File "C:\opt\ros\melodic\x64\lib\site-packages\pkg_resources\__init__.py", line 2443, in load
    return self.resolve()
  File "C:\opt\ros\melodic\x64\lib\site-packages\pkg_resources\__init__.py", line 2449, in resolve
    module = __import__(self.module_name, fromlist=['__name__'], level=0)
  File "C:\opt\ros\melodic\x64\lib\site-packages\vcstool\commands\__init__.py", line 1, in <module>
    from .branch import BranchCommand
  File "C:\opt\ros\melodic\x64\lib\site-packages\vcstool\commands\branch.py", line 6, in <module>
    from .command import Command
  File "C:\opt\ros\melodic\x64\lib\site-packages\vcstool\commands\command.py", line 6, in <module>
    from vcstool.executor import execute_jobs
  File "C:\opt\ros\melodic\x64\lib\site-packages\vcstool\executor.py", line 31
    file=stdout)
SyntaxError: invalid syntax`