ms-iot/ROSOnWindows #381: [Humble] Support MicroROS2 Agent on Windows

ooeygui, 2022-09-14:
ROS Distro: `humble`

**Describe the need**
MicroROS2 is a ROS package for interfacing with Microcontrollers.

**Sources**
``` yaml
repositories:
  micro_ros_msgs:
    type: git
    url: https://github.com/micro-ROS/micro_ros_msgs.git
    version: humble
  micro-ROS-Agent:
    type: git
    url: https://github.com/micro-ROS/micro-ROS-Agent.git
    version:humble
```

**Should this be included in the Microsoft distribution?**
Yes, it is ubiquitous for Microcontrollers.

gavanderhoorn, 2022-09-15:
We've been using the Agent on Windows 10 after building it like this:

```cmd
call "C:\path\to\ros2-humble\local_setup.bat"
mkdir "%USERPROFILE%\agent_ws\src"
git clone -b humble https://github.com/micro-ROS/micro_ros_msgs.git "%USERPROFILE%\agent_ws\src\micro_ros_msgs"
git clone -b humble https://github.com/micro-ROS/micro-ROS-Agent.git "%USERPROFILE%\agent_ws\src\micro-ROS-Agent"
cd "%USERPROFILE%\agent_ws"
colcon build ^
  --merge-install ^
  --packages-up-to micro_ros_agent ^
  --cmake-args ^
  "-DUAGENT_BUILD_EXECUTABLE=OFF" ^
  "-DUAGENT_P2P_PROFILE=OFF" ^
  "--no-warn-unused-cli"
```

No source-code level changes have been necessary so far.

Depending on the exact configuration you'd need, the CMake flags passed to `colcon build` would need to be changed of course.

ooeygui, 2022-09-15:
@gavanderhoorn thanks for the post! I was seeing a build error while working on the humble release; specifically not finding spdlog that it built. Maybe the merge-install took care of it? I was able to work around it by passing the 'use system log` flag.

We do want to include the agent in the official humble release.

ooeygui, 2022-11-28:
Completed.

closed 2022-11-28 by ooeygui

gavanderhoorn, 2022-11-29:
@ooeygui: did you guys upstream any fixes? Or is there a specific fork you're maintaining?

ooeygui, 2022-11-29:
@gavanderhoorn 
For all of humble, I had to make changes to various repos ranging from some simple one liners to lots of new code. We are making haste to announce a 'preview' of a supported LTS of Humble for Windows; where all of the subsystems (Core ROS, Nav2, MoveIt2, MicroROS2) have been exercised. Once we verify stability of the changes in the staging forks, I'll be doing a PR for every change and removing our forks. 

To clarify that point - We do not want *any forks* for ROS.

For transparency - Our build definition for Humble is located here - https://github.com/ms-iot/ros-windows-build/tree/master/ros/humble. Anything with an `ms-iot` is a staging fork. 

We have and will be offering paid support for our distributions through http://aka.ms/ros/support. 

Thank you for your continued support for ROS on Windows!

(Of course, being open source, you can install it now via `choco install ros-humble-desktop --pre`)