ms-iot/ROSOnWindows #384: No gazebo directory in the ROS2 Humble install folder after installing

andrestoga, 2022-11-11:
**Describe the bug**
With a default humble chocolatey install, the gazebo directory is not listed in the ROS2 install folder

**To Reproduce**
Steps to reproduce the behavior:
mkdir c:\opt\chocolatey
set ChocolateyInstall=c:\opt\chocolatey
choco source add -n=ros-win -s="https://aka.ms/ros/public" --priority=1
choco upgrade ros-humble-desktop -y --execution-timeout=0 --pre

**Expected behavior**
I'm expecting to see the following folder in the install path: c:\opt\ros\humble\x64\share\gazebo\setup.bat

**Environment information:**
 - OS version: `Win 10`
 - Chocolatey Versions: `Chocolatey v0.11.3`

ooeygui, 2022-11-14:
Classic Gazebo (Which has been deprecated by Open Robotics) is not part of the ROS2 Humble for Windows distribution. However, Modern Gazebo (specifically Fortress LTS) is included.

andrestoga, 2022-11-14:
Thanks @ooeygui !

So, you mean that ROS2 Humble for Windows 10 hasn't officially been released?

IoTDan, 2023-01-23:
Thank you for reporting this. We will be using Gazebo Fortress and forward for ROS2 Humble on Windows

closed 2023-01-23 by IoTDan