ms-iot/ROSOnWindows #385: [Humble] Gazebo Fortress renders incorrectly - "Flat" instead of shaded.

ooeygui, 2022-11-28:
**Describe the bug**
Gazebo Fortress renders "Flat" using Ogre 1.12. Objects should be shaded.
[image]

traversaro, 2023-01-21:
@ooeygui @polyhobbyst I have the memory that you asked me something on this problem in some issue and I forgot to reply to you, is this real or it is actually my immagination? If that instead is true, sorry for the late response, can you point me to the issue? Thanks!

ooeygui, 2023-01-21:
Hi @traversaro Thanks for closing the loop. I had posed on your gazebo bug report about supporting Ogre 1.12.  I've been scratching my head on this one... Turns out it repros on Linux when using Ogre 1.x... So, not a Windows specific issue - but also makes gazebo visually unusable (although works well for physics sim!)

I'm currently working on Ogre 2 (vcpkg ogre-next) with the OpenGL runtime. I recall hitting a problem where it wasn't rendering because it was using a fix function pipeline on Linux. So revisiting that workstream...

closed 2023-01-23 by IoTDan