ms-iot/ROSOnWindows #386: [Humble] Gazebo logs error for ogre on startup.

ooeygui, 2022-11-28:
**Describe the bug**
During Gazebo startup `ign gazebo shapes.sdf` gazebo generates an error:

```cmd
[GUI] [Err] [D:\a\_work\1\s\gz_ws\src\gz-rendering\ogre\src\OgreRenderEngine.cc:468] Unable to load Ogre Plugin[c:\opt\ros\humble\x64\bin/RenderSystem_GL]. Rendering will not be possible.Make sure you have installed OGRE properly.
```
Although this seems to be incorrect as ogre does initialize and render.

ooeygui, 2023-01-10:
This looks like a bug in gazebo itself. Gazebo doesn't seem to support installation into alternative directories. It embeds the build time directories into the binary itself. So when we relocate the install, we have to use an environment variable. This results in two entries in the ogre path - one which is not necessarily correct, and one which is correct on install. Interestingly enough, it may attempt to install *both* if we set it correctly... Which would then fail....

ooeygui, 2023-01-10:
Sadly, this will impact alternative install locations, so must be fixed. Since we need to set the ogre resource to the rviz directory in the environment, if you actually install it into c:\opt, gazebo will attempt to load plugins will load twice.

Note this line - https://github.com/ms-iot/ros-windows-build/blob/3d11736ea557d5a9300bcd85840d435e6d615f10/ros/humble/src/msft_ros2_env/env-hooks/gazebo.dsv.in#L11 

is wrong - it should point to the vendor_rviz2_ogre directory

traversaro, 2023-01-10:
Not sure if https://github.com/gazebosim/gz-sim/issues/626 is related.

ooeygui, 2023-01-11:
Thank you for connecting this bug to that one! Yes, that is exactly the issue.