ms-iot/ROSOnWindows #390: [Humble] Xacro is not directly callable

polyhobbyist, 2022-11-30:
**Describe the bug**
Xacro is installed with Humble, but just the raw python script without .py in `c:\opt\ros\humble\x64\bin\xacro`. This means it cannot be called from vscode or from launch files.

**To Reproduce**
Open a command Window.
run `xacro`
```cmd
'xacro' is not recognized as an internal or external command,
operable program or batch file.
```

run `python python c:\opt\ros\humble\x64\bin\xacro`
```cmd
Usage: xacro [options] <input>

xacro: error: expected exactly one input file as argument
```

**Expected behavior**
running `xacro` should output help text. Running xacro <urdf.xacro> should output a urdf.

Suggest implementing xacro.bat to call python.

ooeygui, 2022-12-08:
Added https://github.com/ms-iot/ros-windows-build/commit/5d461829742eaeb88715a9316447221b8d94b4af

ooeygui, 2023-01-05:
Reopening because xacro.bat doesn't work with ROS2 launch files. Needs to be a python exe.