ms-iot/ROSOnWindows #395: I'm running ros and qt at the same time, there is no error in the compilation part. But when running it gives an error

serkanMzlm, 2022-12-26:
**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:
* Catkin workspace setup instructions
  * Clone locations
  * build command line
  * Launch command line

**Expected behavior**
A clear and concise description of what you expected to happen.

**Environment information:**
 - OS version: `winver`
 - Chocolatey Versions: `choco list --local-only`

I tried to put ROS2 and QT in the same project and I did it in ubuntu OS. but when i switch this system to windows side i got errors I did the steps on the official page to install ROS windows, but it keeps giving an error. While running talker and listener it doesn't give an error, but when I build my own package colcon it gives an error. When I run the package using ros2 run, I don't get an error, but when I run the .exe extension it created in the buil section, I get the error in the picture. When I compile the code with Qt, I still don't get an error, but when I run the file, I get an error.

Microsoft Visual C++ Runtime Library Debug Error!

...... talker.exe abort() has been called
 (press return to debug the application)

serkanMzlm, 2022-12-27:
I think I understand the cause of the error.  The file created when compiling with colcon build is created in the Release folder. occurs in debug folder when I compile with cmake. There are two files in the Debug folder, the first is exe and the second is pdb extension. How can I make it compile with cmake? Guess if I do this the problem will be solved

serkanMzlm, 2022-12-27:
I converted the Debug folder to Release, for this I entered the following codes and the exe form worked
`
cmake ..
cmake --build . --config Release
`

serkanMzlm, 2022-12-27:
now the problem is how can i write this in cmake file

ooeygui, 2022-12-27:
Thank you for the bug report and investigation.
As you had confirmed, the ROS on Windows only has release binaries.
To solve this in your makefile, you can use MAP_IMPORTED_CONFIG_RELWITHDEBINFO "Release". Here's an example of how it is done in OpenCV https://github.com/opencv/opencv/pull/9656/files

ooeygui, 2022-12-27:
Tagging for Doc - looks like this should be in the porting cookbook - https://ms-iot.github.io/ROSOnWindows/Porting/Cookbook.html

serkanMzlm, 2022-12-28:
> Doküman için etiketleme; Bununla ilgili bilgileri taşıma yemek kitabına ekleyeceğim - https://ms-iot.github.io/ROSOnWindows/Porting/Cookbook.html

Thanks for your help