ms-iot/ROSOnWindows #402: mavros-msgs package

elgarbe, 2023-01-11:
ROS Distro: `melodic`

**Describe the need**
I need mavros package, or at least mavros-msgs in order to visualize on windows computer

**Sources**
Is this a single repository or a collection?
Single: <[source link](https://github.com/mavlink/mavros)>

**Should this be included in the Microsoft distribution?**
Should this or these packages be included in a Microsoft supported distribution?

ooeygui, 2023-01-11:
Do you need all of Mavros or just the messages package?
Context:
I had started working on it a couple years ago, but was blocked because tip of tree for of mavros didn't work together on linux. (it was an odd combination). I haven't looked into it for a bit, but it is on the backlog.

elgarbe, 2023-01-11:
It could be good to have all mavros package, but I know that it will be very difficult. So messages package is ok for me.
Right now I clone mavros repo and copy/paste mavros_msgs floder into my workspace and compile it without a problem. I can echo mavros messages now.

ooeygui, 2023-01-11:
Yep, I'm it :-)

I looked at mavros_msgs; it is part of the mav-ros repo. It can be cloned into a workspace locally, then create an empty file called COLCON_IGNORE in the other source repos to prevent them from being built.

Are you locked to Melodic or can you use Noetic?

ooeygui, 2023-01-11:
(ooeygui is my work associated githb, polyhobbyist is my personal github account)