ms-iot/ROSOnWindows #403: Cannot install ROS 2 on Windows 10: Unable to connect to remote server

iamdejan, 2023-01-16:
**Describe the bug**

I tried to install ROS 2 with [this guide](https://ms-iot.github.io/ROSOnWindows/GettingStarted/SetupRos2.html), but the installation failed.

Log:
```
C:\Windows\System32>choco upgrade ros-foxy-desktop -y --execution-timeout=0 --pre
Chocolatey v1.2.1
Upgrading the following packages:
ros-foxy-desktop
By upgrading, you accept licenses for the packages.
ros-foxy-desktop is not installed. Installing...
ros-foxy-desktop not installed. An error occurred during installation:
 Unable to connect to the remote server
ros-foxy-desktop package files upgrade failed with exit code 1. Performing other installation steps.
The upgrade of ros-foxy-desktop was NOT successful.
ros-foxy-desktop not installed. An error occurred during installation:
 Unable to connect to the remote server

Chocolatey upgraded 0/1 packages. 1 packages failed.
 See the log for details (c:\opt\chocolatey\logs\chocolatey.log).

Failures
 - ros-foxy-desktop (exited 1) - ros-foxy-desktop not installed. An error occurred during installation:
 Unable to connect to the remote server
```

**To Reproduce**

1) Run `x64 Native Tools Command Prompt for VS 2019` as administrator.
2) `mkdir c:\opt\chocolatey`
3) `set ChocolateyInstall=c:\opt\chocolatey`
4) `choco source add -n=ros-win -s="https://aka.ms/ros/public" --priority=1`
5) `choco upgrade ros-foxy-desktop -y --execution-timeout=0 --pre` <- this will produced attached error.

**Expected behavior**
ROS starts downloading.

**Environment information:**
 - OS version: `Windows 10 Version 22H2 (Build 19045.2364)`
 - Chocolatey Version: `v1.2.1`

iamdejan, 2023-01-16:
It seems an issue on my internet service provider. I will close this ticket.

closed 2023-01-16 by iamdejan

ooeygui, 2023-01-17:
Thank you for closing the loop - I'm glad it was resolved!