ms-iot/ROSOnWindows #440: ROS2 Foxy - colcon build C++ with Visual Studio 2022 - change CMake version

anion0278, 2023-02-07:
We are setting up ROS2 in our labs, with Visual Studio 2022 already installed. I try to build Cpp examples with colcon, however, it seems like the version of Cmake that is provided along with Foxy does not yet know the VS2022, so the build fails.  I know that according to the official docs, only VS up to 2019 are supported, however in our case it would be beneficial to avoid installing another VS.
ROS2 Foxy was installed from binaries with Chocolately according to: https://ms-iot.github.io/ROSOnWindows/GettingStarted/SetupRos2.html
Could please suggest any option (colcon argument or env variable) to change the Cmake.exe path during the colcon build?

_My steps:_
The initial error pointed me to the C:\opt\ros\foxy\x64\Lib\site-packages\colcon_cmake\task\cmake\build.py script:
```
…
--- stderr: cpp_pubsub
Traceback (most recent call last):
  File "c:\opt\ros\foxy\x64\lib\site-packages\colcon_core\executor\__init__.py", line 91, in __call__
    rc = await self.task(*args, **kwargs)
  File "c:\opt\ros\foxy\x64\lib\site-packages\colcon_core\task\__init__.py", line 93, in __call__
    return await task_method(*args, **kwargs)
  File "c:\opt\ros\foxy\x64\lib\site-packages\colcon_ros\task\ament_cmake\build.py", line 59, in build
    rc = await extension.build(
  File "c:\opt\ros\foxy\x64\lib\site-packages\colcon_cmake\task\cmake\build.py", line 87, in build
    rc = await self._reconfigure(args, env)
  File "c:\opt\ros\foxy\x64\lib\site-packages\colcon_cmake\task\cmake\build.py", line 163, in _reconfigure
    raise RuntimeError(
RuntimeError: Unknown / unsupported VS version '17.0'
…
```

Which was easily "fixed" by appending '17.0': 'Visual Studio 17 2022' to the dictionary:
```
supported_vsv = {
                '17.0': 'Visual Studio 17 2022',
                '16.0': 'Visual Studio 16 2019',
                '15.0': 'Visual Studio 15 2017',
                '14.0': 'Visual Studio 14 2015',
            }
```

A new attempt to build now suggests that the Cmake provided with the Foxy does not recognize the VS 2022 C++ compiler:
```
…
--- stderr: cpp_pubsub
CMake Error: Could not create named generator Visual Studio 17 2022

Generators
  Visual Studio 16 2019        = Generates Visual Studio 2019 project files.
                                 Use -A option to specify architecture.
  Visual Studio 15 2017 [arch] = Generates Visual Studio 2017 project files.
                                 Optional [arch] can be "Win64" or "ARM".
  Visual Studio 14 2015 [arch] = Generates Visual Studio 2015 project files.
                                 Optional [arch] can be "Win64" or "ARM".
  Visual Studio 12 2013 [arch] = Generates Visual Studio 2013 project files.
                                 Optional [arch] can be "Win64" or "ARM".
  Visual Studio 11 2012 [arch] = Generates Visual Studio 2012 project files.
                                 Optional [arch] can be "Win64" or "ARM".
  Visual Studio 10 2010 [arch] = Generates Visual Studio 2010 project files.
                                 Optional [arch] can be "Win64" or "IA64".
  Visual Studio 9 2008 [arch]  = Generates Visual Studio 2008 project files.
                                 Optional [arch] can be "Win64" or "IA64".
  Borland Makefiles            = Generates Borland makefiles.
…
```

By checking the version _C:\opt\ros\foxy\x64\Scripts\cmake.exe --version_ I could see that the default vesion of CMake is 3.18.2. I hoped, that by installing a newer version of Cmake (3.25.2) and by prepending the PATH env var with its corresponding path (C:\Program Files\CMake\bin) would be sufficient to force Colcon to utilize the different version of Cmake. But it wasn't :) .

Could please suggest any option (colcon argument or env variable) to change the Cmake.exe path during the colcon build?

anion0278, 2023-02-07:
I guess I have found a solution, though it was so close that I should have checked it before :/ .
Looking at C:\opt\ros\foxy\x64\Lib\site-packages\colcon_cmake\task\cmake\__init__.py pointed me in the right direction. 

```
"""Environment variable to override the CMake executable"""
CMAKE_COMMAND_ENVIRONMENT_VARIABLE = EnvironmentVariable(
    'CMAKE_COMMAND', 'The full path to the CMake executable')
```

So by setting env var CMAKE_COMMAND I could change the path to CMake executable:
```
set CMAKE_COMMAND=C:/Program Files/CMake/bin/cmake.exe
```

colcon now succesfully builds my Cpp packages. 
At least I hope this information might be of use to someone else :) .
(I guess its partially related to #356 )

I wonder if it is possible to update the build configuration to incorporate a newer version of CMake into Foxy distribution?

gavanderhoorn, 2023-02-07:
`CMAKE_COMMAND` is also mentioned in the `colcon` usage output:

```
colcon --help
usage: colcon [-h] [--log-base LOG_BASE] [--log-level LOG_LEVEL]
              {build,extension-points,extensions,graph,info,list,metadata,mixin,test,test-result,version-check} ...

options:
...

colcon verbs:
...

Environment variables:
  CMAKE_COMMAND         The full path to the CMake executable
  ...
```

anion0278, 2023-02-07:
:facepalm: I should have read this more carefully :D . thanks!
However, I wonder if it is possible to update the build configuration to incorporate a newer version of CMake into Foxy distribution so that it works out of the box for everyone?

ooeygui, 2023-02-07:
@anion0278 ROS on Windows has a toolchain affinity, instead of ROS on Linux which has a distro affinity. Our binaries are compiled with the default toolchain for Visual Studio 2019, so it is required to be installed to build nodes for Noetic and Foxy. We can't change the toolchain for foxy; otherwise we would break existing deployments.

With Humble, we are moving to VS2022. Both can be installed in parallel.

ooeygui, 2023-02-07:
Wrong button. Leaving open for consideration since installing VS2019 is getting difficult.

anion0278, 2023-06-08:
I'd like to mention that if anyone intends to utilize the VS 22 compiler with a newer version of CMake, it's crucial to select versions prior to 3.24. Specifically, I would recommend using version 3.23.3 (refer to the related issue: https://github.com/dusty-nv/jetson-containers/issues/181#issuecomment-1216802257). The newer versions of CMake introduce ament-related issues when building Turtlebot3 from source (`Failed to find exported target names`). Alternatively, it seems to be possible to apply a fix for `ament`: https://github.com/ros-planning/moveit2/issues/1570#issuecomment-1272390429. 
To avoid confusion, I have edited CMake version in my initial message.