ms-iot/ROSOnWindows #441: [Melodic, Noetic] Default PyAssimp version 4.1.4 causes silent crash when loading STL through MoveIt's PlanningSceneInterface

anion0278, 2023-02-13:
**Describe the bug**
PyAssimp version 4.1.4 causes a silent crash when loading STL through MoveIt's PlanningSceneInterface (Python).
The issue can be fixed by installing pyAssimp==4.1.3. The issue is present in both Melodic and Noetic and seems to be rather old (https://github.com/ros-planning/moveit/issues/86#issuecomment-480698001, https://answers.ros.org/question/316515/cannot-add-mesh-files-to-moveit/) and related to pyAssimp itself rather than ROS (https://github.com/assimp/assimp/issues/2343) .
A potential solution is to change the installed version in  [patch.bat](https://github.com/ms-iot/ros-windows-build/blob/fdf539657d807e9cbae3fc3297098361bf95cce8/ros/noetic/patch.bat#L21)

**To Reproduce**
Please see the attached MRE: [example.zip](https://github.com/ms-iot/ROSOnWindows/files/10719985/example.zip)
Steps:
```
roslaunch panda_moveit_config demo.launch
rosrun panda_moveit_config src/attached_obj_collision.py
```
`attached_obj_collision.py` silently fails after printing `MoveIt commander ok`, objects are not added into the scene, trajectory is not planned.

**Expected behavior**
Running `attached_obj_collision.py` (violet object in the attached image) should load `work_object.stl` into the planning scene and execute the trajectory around two green boxes.
[image]

**Environment information:**
 - OS version: Win11 22H2 22621.1105
 - Chocolatey Versions:
```
Chocolatey v1.2.0
chocolatey-compatibility.extension 1.0.0
chocolatey-core.extension 1.4.0
chocolatey-windowsupdate.extension 1.0.5
KB2919355 1.0.20160915
KB2919442 1.0.20160915
KB2999226 1.0.20181019
KB3033929 1.0.5
KB3035131 1.0.3
ros-melodic-moveit 20201221.0.0.2104300002
vcredist140 14.34.31931
vcredist2010 10.0.40219.32503
11 packages installed.
```

ooeygui, 2023-03-09:
Thanks for the report. Given the upstream issue, I'm tagging on all distros to validate.

songyuc, 2023-07-25:
Hi! Is ROS2-Iron supported on Win11?

songyuc, 2023-07-25:
The doc said they only support [Win10](https://docs.ros.org/en/iron/Installation/Windows-Install-Binary.html#system-requirements).