ms-iot/ROSOnWindows #448: [Foxy] colcon build returns PermissionError WinError5 when trying to build any package.

VinHub-Gitcent, 2023-10-07:
I am experiencing problems with colcon build. I am running it in x64 Native Tools Command Prompt for VS 2019 as admin. I get this error for every colcon build i was trying to create.

I am using ROS2 foxy for Windows. View the full Error Message below:

```
C:\dev\ros2_ws>colcon build --merge-install
[1.895s] root DEBUG Using proactor: IocpProactor
Starting >>> turtlesim
[1.911s] colcon.colcon_ros.prefix_path.catkin WARNING The path 'C:\opt\ros\foxy\x64\tools\vcpkg\installed\x64-windows' in the environment variable CMAKE_PREFIX_PATH doesn't exist
['"C:\\dev\\ros2_ws\\build\\turtlesim\\colcon_command_prefix_build.bat" && set']
--- stderr: turtlesim
Traceback (most recent call last):
  File "c:\opt\ros\foxy\x64\lib\site-packages\colcon_core\executor\__init__.py", line 91, in __call__
    rc = await self.task(*args, **kwargs)
  File "c:\opt\ros\foxy\x64\lib\site-packages\colcon_core\task\__init__.py", line 93, in __call__
    return await task_method(*args, **kwargs)
  File "c:\opt\ros\foxy\x64\lib\site-packages\colcon_ros\task\ament_cmake\build.py", line 59, in build
    rc = await extension.build(
  File "c:\opt\ros\foxy\x64\lib\site-packages\colcon_cmake\task\cmake\build.py", line 87, in build
    rc = await self._reconfigure(args, env)
  File "c:\opt\ros\foxy\x64\lib\site-packages\colcon_cmake\task\cmake\build.py", line 174, in _reconfigure
    completed = await run(
  File "c:\opt\ros\foxy\x64\lib\site-packages\colcon_core\task\__init__.py", line 177, in run
    completed = await colcon_core_subprocess_run(
  File "c:\opt\ros\foxy\x64\lib\site-packages\colcon_core\subprocess.py", line 81, in run
    rc, _, _ = await _async_check_call(
  File "c:\opt\ros\foxy\x64\lib\site-packages\colcon_core\subprocess.py", line 140, in _async_check_call
    process = await create_subprocess(
  File "asyncio\subprocess.py", line 236, in create_subprocess_exec
  File "asyncio\base_events.py", line 1630, in subprocess_exec
  File "asyncio\windows_events.py", line 385, in _make_subprocess_transport
  File "asyncio\base_subprocess.py", line 36, in __init__
  File "asyncio\windows_events.py", line 881, in _start
  File "asyncio\windows_utils.py", line 153, in __init__
  File "subprocess.py", line 854, in __init__
  File "subprocess.py", line 1307, in _execute_child
PermissionError: [WinError 5] Access is denied
---
Failed   <<< turtlesim [0.16s, exited with code 1]

Summary: 0 packages finished [0.42s]
  1 package failed: turtlesim
  1 package had stderr output: turtlesim
```

  
This is a really basic operation and should run out of the box when following the installation guide from:
https://ms-iot.github.io/ROSOnWindows/GettingStarted/SetupRos2.html

Does Anyone has an idea why this error occures?

ooeygui, 2023-10-09:
Hi there,
I have not seen this specific issue before. I suspect that this may be blocked by a virus scanner?

VinHub-Gitcent, 2023-10-10:
Hi thank you for your reply. I think the virus scanner is not interfering here. 
I wonder if the Warning:  
`WARNING The path 'C:\opt\ros\foxy\x64\tools\vcpkg\installed\x64-windows' in the environment variable CMAKE_PREFIX_PATH doesn't exist` should occur here.
This is what I got from the colcon log file, maybe this helps.

```
[0.572s] [34mcolcon[0m [1;30mDEBUG[0m [32mCommand line arguments: ['c:\\opt\\ros\\foxy\\x64\\Scripts\\colcon', 'build', '--merge-install', '--event-handlers', 'console_direct+'][0m
[0.572s] [34mcolcon[0m [1;30mDEBUG[0m [32mParsed command line arguments: Namespace(ament_cmake_args=None, base_paths=['.'], build_base='build', catkin_cmake_args=None, catkin_skip_building_tests=False, cmake_args=None, cmake_clean_cache=False, cmake_clean_first=False, cmake_force_configure=False, cmake_target=None, cmake_target_skip_unavailable=False, continue_on_error=False, event_handlers=['console_direct+'], executor='parallel', ignore_user_meta=False, install_base='install', log_base=None, log_level=None, main=<bound method BuildVerb.main of <colcon_core.verb.build.BuildVerb object at 0x000000000406D1C0>>, merge_install=True, metas=['./colcon.meta'], packages_above=None, packages_above_and_dependencies=None, packages_above_depth=None, packages_end=None, packages_ignore=None, packages_ignore_regex=None, packages_select=None, packages_select_build_failed=False, packages_select_by_dep=None, packages_select_regex=None, packages_select_test_failures=False, packages_skip=None, packages_skip_build_finished=False, packages_skip_by_dep=None, packages_skip_regex=None, packages_skip_test_passed=False, packages_skip_up_to=None, packages_start=None, packages_up_to=None, packages_up_to_regex=None, parallel_workers=12, paths=None, symlink_install=False, test_result_base=None, verb_extension=<colcon_core.verb.build.BuildVerb object at 0x000000000406D1C0>, verb_name='build', verb_parser=<colcon_defaults.argument_parser.defaults.DefaultArgumentsDecorator object at 0x000000000406D670>)[0m
[0.625s] [34mcolcon.colcon_core.package_discovery[0m [1;30mLevel 1[0m discover_packages(colcon_meta) check parameters
[0.625s] [34mcolcon.colcon_core.package_discovery[0m [1;30mLevel 1[0m discover_packages(recursive) check parameters
[0.625s] [34mcolcon.colcon_core.package_discovery[0m [1;30mLevel 1[0m discover_packages(ignore) check parameters
[0.625s] [34mcolcon.colcon_core.package_discovery[0m [1;30mLevel 1[0m discover_packages(path) check parameters
[0.625s] [34mcolcon.colcon_core.package_discovery[0m [1;30mLevel 1[0m discover_packages(colcon_meta) discover
[0.625s] [34mcolcon.colcon_core.package_discovery[0m [1;30mLevel 1[0m discover_packages(recursive) discover
[0.625s] [34mcolcon.colcon_core.package_discovery[0m [1;30mINFO[0m Crawling recursively for packages in 'C:\dev\ros2_ws'
[0.625s] [34mcolcon.colcon_core.package_identification[0m [1;30mLevel 1[0m _identify(.) by extensions ['ignore', 'ignore_ament_install']
[0.625s] [34mcolcon.colcon_core.package_identification[0m [1;30mLevel 1[0m _identify(.) by extension 'ignore'
[0.625s] [34mcolcon.colcon_core.package_identification[0m [1;30mLevel 1[0m _identify(.) by extension 'ignore_ament_install'
[0.625s] [34mcolcon.colcon_core.package_identification[0m [1;30mLevel 1[0m _identify(.) by extensions ['colcon_pkg']
[0.625s] [34mcolcon.colcon_core.package_identification[0m [1;30mLevel 1[0m _identify(.) by extension 'colcon_pkg'
[0.625s] [34mcolcon.colcon_core.package_identification[0m [1;30mLevel 1[0m _identify(.) by extensions ['colcon_meta']
[0.625s] [34mcolcon.colcon_core.package_identification[0m [1;30mLevel 1[0m _identify(.) by extension 'colcon_meta'
[0.625s] [34mcolcon.colcon_core.package_identification[0m [1;30mLevel 1[0m _identify(.) by extensions ['ros']
[0.625s] [34mcolcon.colcon_core.package_identification[0m [1;30mLevel 1[0m _identify(.) by extension 'ros'
[0.640s] [34mcolcon.colcon_core.package_identification[0m [1;30mLevel 1[0m _identify(.) by extensions ['cmake', 'python']
[0.640s] [34mcolcon.colcon_core.package_identification[0m [1;30mLevel 1[0m _identify(.) by extension 'cmake'
[0.640s] [34mcolcon.colcon_core.package_identification[0m [1;30mLevel 1[0m _identify(.) by extension 'python'
[0.640s] [34mcolcon.colcon_core.package_identification[0m [1;30mLevel 1[0m _identify(.) by extensions ['python_setup_py']
[0.640s] [34mcolcon.colcon_core.package_identification[0m [1;30mLevel 1[0m _identify(.) by extension 'python_setup_py'
[0.640s] [34mcolcon.colcon_core.package_identification[0m [1;30mLevel 1[0m _identify(build) by extensions ['ignore', 'ignore_ament_install']
[0.640s] [34mcolcon.colcon_core.package_identification[0m [1;30mLevel 1[0m _identify(build) by extension 'ignore'
[0.640s] [34mcolcon.colcon_core.package_identification[0m [1;30mLevel 1[0m _identify(build) ignored
[0.640s] [34mcolcon.colcon_core.package_identification[0m [1;30mLevel 1[0m _identify(install) by extensions ['ignore', 'ignore_ament_install']
[0.640s] [34mcolcon.colcon_core.package_identification[0m [1;30mLevel 1[0m _identify(install) by extension 'ignore'
[0.640s] [34mcolcon.colcon_core.package_identification[0m [1;30mLevel 1[0m _identify(install) ignored
[0.640s] [34mcolcon.colcon_core.package_identification[0m [1;30mLevel 1[0m _identify(log) by extensions ['ignore', 'ignore_ament_install']
[... cut ...]
[5.884s] [34mcolcon.colcon_core.shell[0m [1;30mINFO[0m Creating prefix chain script 'C:\dev\ros2_ws\install\setup.ps1'
[5.884s] [34mcolcon.colcon_core.shell[0m [1;30mINFO[0m Creating prefix script 'C:\dev\ros2_ws\install\local_setup.bat'
[5.884s] [34mcolcon.colcon_core.shell[0m [1;30mINFO[0m Creating prefix util module 'C:\dev\ros2_ws\install\_local_setup_util_bat.py'
[5.884s] [34mcolcon.colcon_core.shell[0m [1;30mINFO[0m Creating prefix chain script 'C:\dev\ros2_ws\install\setup.bat'
```

What are your thoughts on this?

ooeygui, 2023-10-10:
Thanks for the log. The warning is a red-herring. We preemptively put that in the path, in case a rosdep needs to use vcpkg. If you create that directory, the warning will go away. (We should have done that in setup, but it was broken at some point)

This line is suspect `C:\Program Files\CMake\bin C:\dev\ros2_ws\src\ros_tutorials\turtlesim -DCMAKE_INSTALL_PREFIX=C:\dev\ros2_ws\install -G Visual Studio 16 2019`

It looks like it is missing cmake.exe - it should be:
`C:\Program Files\CMake\bin\cmake.exe C:\dev\ros2_ws\src\ros_tutorials\turtlesim -DCMAKE_INSTALL_PREFIX=C:\dev\ros2_ws\install -G Visual Studio 16 2019

 I won't ask you to post the environment as it includes sensitive info.
 However, I would have expected it to use the one deployed in `c:\opt\ros\foxy\x64\cmake.exe`.
 Perhaps you have an environment overriding it?

VinHub-Gitcent, 2023-10-17:
Thanks for your reply. 
I had an existing installation of CMAKE on my system. After uninstalling it and removing from PATH the colcon command works as intended. 
I still don't get why using another functional installation of Cmake crashes colcon, but the issue can be considered closed I think.

I wonder how I can have a second version of Cmake installed without it interfering with the colcon command of foxy. Maybe you can give me some idea for that. 

Thanks again and best regards.

closed 2023-10-17 by VinHub-Gitcent

ooeygui, 2023-10-17:
Thank you for closing the loop. I suspect that between versions of cmake the environment variable that's used to locate it changed behaviors? Or on Windows the environment variable behaves differently than on linux?

I'm glad it is working.