ms-iot/ROSOnWindows #450: Plans for MS IoT moving to a non EOL Version of ROS2

VinHub-Gitcent, 2023-11-06:
Dear MS-IoT Team,

are there any plans moving away from ROS 2 foxy? Is there a roadmap somewhere for when it's planned to move to e.g. ROS2 humble?
Packages like Moveit are no longer maintained for foxy and new features cannot be used. (e.g. python bindings for Moveit 2)

Do you have some information you could share?

Best Regards

headlee, 2024-01-11:
You should be able to get a pre-release version of ms-iot's humble build by following the steps [here](https://ms-iot.github.io/ROSOnWindows/GettingStarted/SetupRos2.html), but replace the `foxy` with `humble` in the `choco upgrade` step (note the `--pre` is important).

I don't know if it will have everything you need...it doesn't look like any work has been done on their humble build since March 2023. We're not using it because we need one of the later humble patch releases which fixes some bugs with fastrtps.

I'm also interested in knowing about future plans, especially related to Jammy!

ooeygui, 2024-01-11:
Microsoft is no longer providing independent ROS releases and recommends using the binary distributions from Open Robotics.