ms-iot/ROSOnWindows #451: roscore killing on exit

umicar1234, 2024-02-08:
mech@mech-Latitude-3420:~$ roscore
... logging to /home/mech/.ros/log/f7c25a4c-c697-11ee-9aea-89e495e184c7/roslaunch-mech-Latitude-3420-18762.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://mech-Latitude-3420:42703/
ros_comm version 1.16.0

SUMMARY
========

PARAMETERS
 * /rosdistro: noetic
 * /rosversion: 1.16.0

NODES

auto-starting new master
process[master]: started with pid [18770]
ROS_MASTER_URI=http://mech-Latitude-3420:11311/

setting /run_id to f7c25a4c-c697-11ee-9aea-89e495e184c7
Traceback (most recent call last):
  File "/opt/ros/noetic/lib/python3/dist-packages/roslaunch/__init__.py", line 347, in main
    p.start()
  File "/opt/ros/noetic/lib/python3/dist-packages/roslaunch/parent.py", line 316, in start
    self.runner.launch()
  File "/opt/ros/noetic/lib/python3/dist-packages/roslaunch/launch.py", line 676, in launch
    self._setup()
  File "/opt/ros/noetic/lib/python3/dist-packages/roslaunch/launch.py", line 653, in _setup
    self._launch_core_nodes()
  File "/opt/ros/noetic/lib/python3/dist-packages/roslaunch/launch.py", line 530, in _launch_core_nodes
    name, success = self.launch_node(node, core=True)
  File "/opt/ros/noetic/lib/python3/dist-packages/roslaunch/launch.py", line 558, in launch_node
    process = create_node_process(self.run_id, node, master.uri, sigint_timeout=self.sigint_timeout, sigterm_timeout=self.sigterm_timeout)
  File "/opt/ros/noetic/lib/python3/dist-packages/roslaunch/nodeprocess.py", line 154, in create_node_process
    args = create_local_process_args(node, machine)
  File "/opt/ros/noetic/lib/python3/dist-packages/roslaunch/node_args.py", line 264, in create_local_process_args
    matches = roslib.packages.find_node(node.package, node.type, rospack=rospack)
  File "/opt/ros/noetic/lib/python3/dist-packages/roslib/packages.py", line 417, in find_node
    return find_resource(pkg, node_type, filter_fn=_executable_filter, rospack=rospack)
  File "/opt/ros/noetic/lib/python3/dist-packages/roslib/packages.py", line 526, in find_resource
    search_paths = catkin_find(
  File "/opt/ros/noetic/lib/python3/dist-packages/catkin/find_in_workspaces.py", line 149, in find_in_workspaces
    source_path_to_packages[source_path] = find_packages(source_path)
  File "/usr/lib/python3/dist-packages/catkin_pkg/packages.py", line 92, in find_packages
    packages = find_packages_allowing_duplicates(basepath, exclude_paths=exclude_paths,
  File "/usr/lib/python3/dist-packages/catkin_pkg/packages.py", line 155, in find_packages_allowing_duplicates
    parsed_package = parse_package_string(
  File "/usr/lib/python3/dist-packages/catkin_pkg/package.py", line 786, in parse_package_string
    raise InvalidPackage('Error(s):%s' % (''.join(['\n- %s' % e for e in errors])), filename)
catkin_pkg.package.InvalidPackage: Error(s) in package '/home/mech/danger_ws/src/fly_bot/package.xml':
Error(s):
- The manifest of package "fly_bot" (with format version 2) must not contain the following tags: run_depend
- Please replace <run_depend> tags with <exec_depend> tags.

it works perfectly before but suddenly gets stop working

ooeygui, 2024-02-08:
Hi there,
This looks like a linux callstack. I suspect that noetic got an update on your machine. The error seems to be:
```bash
The manifest of package "fly_bot" (with format version 2) must not contain the following tags: run_depend
Please replace <run_depend> tags with <exec_depend> tags.
```

It looks like you need to change the `package.xml` file.

umicar1234, 2024-02-09:
> Hi there, This looks like a linux callstack. I suspect that noetic got an update on your machine. The error seems to be:
> 
> ```shell
> The manifest of package "fly_bot" (with format version 2) must not contain the following tags: run_depend
> Please replace <run_depend> tags with <exec_depend> tags.
> ```
> 
> It looks like you need to change the `package.xml` file.

It works, thanks

umicar1234, 2024-02-09:
I just have to change run_command to exec_command

closed 2024-02-09 by ooeygui