ms-iot/ROSOnWindows #452: Installing ROS 2 Binaries fails with "Invalid credentials specified."

DanielChaseButterfield, 2024-05-14:
**Describe the bug**
I am unable to complete the steps for [Installing ROS 2 Binaries](https://ms-iot.github.io/ROSOnWindows/GettingStarted/SetupRos2.html) as when I run the command `choco upgrade ros-humble-desktop -y --execution-timeout=0 --pre --trace`, I get the following error:

```
Invalid credentials specified.
Please provide credentials for: https://aka.ms/ros/public
User name: 
Password:
```

No matter what credentials I enter, it always fails.

**To Reproduce**
The steps found here: https://ms-iot.github.io/ROSOnWindows/GettingStarted/SetupRos2.html

However, I edited the final command to install humble instead:
```
choco upgrade ros-humble-desktop -y --execution-timeout=0 --pre
```

Another note: If I navigate to the repository `https://aka.ms/ros/public` in the browser, I get a similar access denied error.

[image]

**Expected behavior**
It should install ROS 2 humble as expected.

**Environment information:**
 - OS version: Windows 11 23H2
 - Chocolatey Versions: 

 ```
Chocolatey v2.2.2
0 packages installed.
```

DanielChaseButterfield, 2024-05-14:
Looking through other more recent issues, looks like Microsoft privated their ros repositories: 
https://github.com/ms-iot/ROSOnWindows/issues/450#issuecomment-1887797265 .

Therefore, the documentation should probably be fixed to provide an alternative install.

ooeygui, 2024-05-14:
Hello @DanielChaseButterfield,
Thank you for your interest in ROS on Windows.

As mentioned in that comment, Microsoft has sunset its distribution endpoint of ROS on Windows and recommends using the Windows distributions provided by Open Robotics as documented on https://docs.ros.org/en/humble/Installation/Windows-Install-Binary.html.